Jack000/Deepnest #1: Enhancement suggestion: Required orientation

HJuup, 2018-08-10:
Hi Jack,

Great work has been done as can be seen in your deepnest release.
- Big improvement to be able to import DXF
- Good looking user interface
- Working towards .cc code

Is there any change you will be working on SVGnest issues #15 and  #57 ?
That is about grain directions and orientations. Working towards a solution for that would make SvgNest/DeepNest really stand out from commercially available solutions.

Some other questions:
_ While nesting the resultbar shows all kinds of colors, what is the meaning of these colors, what can a user learn from that?
- The register/login section option: What is the puropse apart form being able to do an export?

Appreciate your efforts put in SVGnest/Deepnest !

jnweiger, 2018-10-06:
Issues 15 and 57 seem to be something different now. Plesse doublecheck these references.

Preserving grain orientation should be a setting per part, if that is possible. 
For some parts I need the grain to go lengthwise for mechanical strength, of other parts it may be irrelevant.

silopolis, 2018-10-12:
👍

Super bonus point if you can select several parts and set the grain
continuous accross them

>