Jack000/Deepnest #2: Deepest using C++ simple console app

petrasvestartas, 2018-08-10:
Hi

Is there any way or guides using deepest in simple cpp console application or it is based on java script?

Jack000, 2018-08-11:
for nfp generation you can just follow the tutorial here: https://www.boost.org/doc/libs/1_67_0/libs/polygon/doc/gtl_minkowski_tutorial.htm

the nfp is the minkowski difference between two polygons

petrasvestartas, 2018-08-11:
Thank you for the reference.

I would like to ask a bit clarification of minkowski sums. 

I thought that minkowski sum and no fit polygon are two different things. Am I wrong?

When I Google minkowski sums in one reference polygons are dragged around main polygon within a centre:
https://doc.cgal.org/4.5/Minkowski_sum_2/tight.png

But the no fit polygon is not moving within centre but by corner: http://www.cs.cmu.edu/~motionplanning/student_gallery/2007/Hyungsoo/HW5/Motion_Planning_HW5_files/1.jpg

Are these ones two different algorithms and the one you reference is the later?

Jack000, 2018-08-11:
if you take the minkowski difference between two polygons the result is the nfp. The reference point on the second polygon is arbitrary, it just represents a constant offset of the nfp and has no impact on its shape.

petrasvestartas, 2018-08-12:
Thank you for reply.
I am using clipper library for now and I have a possibility to compute Minkowski sum and subtract two polygons using the same clipper library.

The other question I have: what is the method conceptually the simplest case how you place B element in nfp? 
Do you compute all possible Minkowski sums per pair A-B?
For instance if you polygon have 5 vertices then you compute 5 minkowski sums?
Or you just check at 1 vertex?

This is the question for me, because in case of computer Minkowski sum for 1 vertex, some areas in polygon clearly fits and in areas do not. Please look at screenshot attached.

[image]

Jack000, 2018-08-12:
hm.. the minkowski sum is only used to compute the nfp, not for choosing vertices. Once the nfp is computed the choice of which vertex to put B is done heuristically - either the left-most (for center of gravity method), lowest bounding box, or smallest convex hull.

petrasvestartas, 2018-08-12:
Sorry maybe I did not formulate my question clearly it is a bit more simple.

 Before getting into heuristics. 
Do you check for collisions for placing elements inside nfp?

Like this scheme below, how do you know that element B will not intersect A?
[image]

Jack000, 2018-08-12:
the nfp is the polygon that defines the boundary between collision and non-collision. ie - at any vertex of the nfp A and B are touching but not colliding with eachother. Once the nfp has been found the collision problem has already been solved, and it's just a matter of picking which vertex to use.

petrasvestartas, 2018-08-13:
Ah, it seems I mixed two different things minkowski **sum** and minkowski **difference**.
In your screenshots it was minkowski difference.

Just to know, what is minkowski sum is typically used for?

[image]

Jack000, 2018-08-13:
they're very similar concepts. The minkowski difference is simply the minkowski sum of two polygons, where one of the polygons is inverted (I forget A or B, it's been a while since I've looked at this stuff : )

vupqtran, 2018-08-18:
Hi @Jack000  and @petrasvestartas,
Can you help me to generate visual studio project files for this project?
Thanks a lot.

Jack000, 2018-08-18:
hm.. it's an electron project so it's built using electron-builder. I'm not sure how it would work with visual studio but if you're just using the c code I assume you can just copy/paste the file, it's not very big.

vupqtran, 2018-08-18:
Hi @Jack000 ,
I see this can help me generate visual studio project from "gyp project"
https://github.com/mogemimi/gyp-getting-started

Jack000, 2018-08-18:
I think that's just a part of the build process though, I'm not sure it ends up building usable VS projects.

Jack000, 2018-08-20:
ah ok, glad it worked out

closed 2018-08-20 by Jack000

BartekPacocha, 2018-09-24:
Hello, @vupqtran 
Could you give me some help with generate visual studio project files? I've try that link from you and i still have some errors. Could you give me step by step instructions?

vupqtran, 2018-09-26:
> Hello, @vupqtran
> Could you give me some help with generate visual studio project files? I've try that link from you and i still have some errors. Could you give me step by step instructions?

Please, give me your errors's detail. Hope can i help you.

feiyuefei, 2020-06-16:
Hi, @vupqtran
Could you tell me some details or steps to generate visual studio project files? I am trying  to generate visual studio project files by employing gyp.I still have a lot of confusion about it.I just use windows and visual studio 2013.Looking forward to contacting you further.
Thanks a lot.