Jack000/Deepnest #3: Some lines incorrect in SVG export

marmilicious, 2018-08-23:
Hello.  I just tried Deepnest out today and think it's very cool!
I ran into an issue where some of the lines seem to be incorrect in an exported SVG file (even though it looks correct in the the Deepnest preview before exporting).

I made a part in CAD (using Onshape), and as part of the design the part is mirrored across a line on a 45deg angle, and then those parts were again mirrored across another plane.  From CAD I exported a drawing as DXF 2013 and also tried as DXF R14.  Both of these import into Deepnest just fine, and once nested look correct in the preview.  After exporting as SVG and opening in Inkscape two parts have incorrect lines though.
https://i.imgur.com/YafrgCI.jpg

Here are the 2013 and R14 DXF files if that's any use.
https://pastebin.com/Jtd4ue4u
https://pastebin.com/G8MZ1b9E

If I import these DXF files into Inkscape all parts look correct.  Maybe something about the mirroring caused this?
Thank you
-marc

Jack000, 2018-08-23:
hm.. could you send me the exported svg? Does the resulting svg render correctly in a browser? My guess is that the mirroring is causing problems with the arc commands.

marmilicious, 2018-08-23:
@Jack000  
Here's the SVG file.  
https://pastebin.com/vftF4Zvh

Bringing that back into Deepnest shows the same couple of incorrect lines as seen in Inkscape.

marmilicious, 2018-08-23:
Sorry, I missed where you said "brower".  No, it's incorrect there as well.

Jack000, 2018-08-23:
I'm actually having some trouble importing the dxf files.

- Does it break if you convert the dxf into svg in inkscape, then import the svg?
- Does it still happen if you disable the merge lines feature in config?

marmilicious, 2018-08-24:
DXF files should be in mm.

Test 1)
I imported the DXF (2013 verison) into to Inkscape (looks correct) and saved as SVG.
Imported that SVG into Deepnest (looks correct) and ran nesting.
Exported as SVG and viewed this new file in a browser and those lines are still incorrect.

Test 2)
Turned Merge common lines off, ran nesting again, exported and... it looks correct now!

Test 3)
I also went back and tested importing the DXF (2013 version) again, and ran nesting with Merge common lines off, and this also looks correct now.  So turning that off seems to be a fix for now.

Thank you for your help with this.
-marc

marmilicious, 2018-08-24:
Tested and confirmed.  Super.  Thank you!

closed 2018-08-24 by marmilicious