Jack000/Deepnest #4: Help to build Deepnest app for window

vupqtran, 2018-08-24:
Hi @Jack000 ,
Can you share me how to build deepnest app for window?
I clone your source code and using npm to build but not success. It build complete but when run it can't start nest.
Thanks.

huyhl248, 2018-08-27:
Hi @Jack000,
"the problem is the native addon", please give me more detail? I have same issue, no error when building. I can import, select object, 
[image]
but when I click on "Start nest", the software did not do anything.
[image]
Please see the attached images.
Could you tell me what is the part of code that could effect? 
Thanks a lot!

Jack000, 2018-08-27:
this means the c++ addon that's used to calculate the minkowski sum is not working

the addon is located in the minkowski directory. To see what the error is, unhide the web worker window ( show: true ) and open its developer tools - these configs are located in main.js

Jack000, 2018-08-30:
I think you need to build the addon on your local system. Try npm clean and npm build, then copy the files from ./build to ./minkowski

vupqtran, 2018-08-31:
Hi @Jack000 ,
Can you tell me your script to build deepnest source code using npm? 
ex:  
npm run configure
npm run build
....
Thanks.

vupqtran, 2018-08-31:
Hi @Jack000 ,
I build successfully for **windows 7 64 bit**. But in your new content of package.json, i change this line:
**"postinstall": "npm i electron@1.4.8 --arch=ia32 --save-dev"**
and run postinstall before run install.
Thanks a lot.

Jack000, 2018-08-31:
ok great, let me know if you have any other questions

closed 2018-08-31 by vupqtran