Jack000/Deepnest #5: Request to recognize and color interior holes in parts

marmilicious, 2018-08-24:
Typically in laser cutting we want to cut out all the interior lines (holes in a part) before cutting around the perimeter of a part.  When a part is cut free from the sheet material sometimes it's position shifts a little bit.  If holes are cut second then they can be slightly misaligned from their correct location.

[image]

SVG file of above image:   https://pastebin.com/4dfXWb0c

If Deepnest could recognize interior lines (holes) and color these with a different stroke color when exporting SVG files then it would make if very easy to select these (select all lines with same stroke color), and be able to tell the laser cutter to cut those first.  Ideally there would be an option to set the stroke color for interior and exterior lines.

deece, 2018-11-30:
What laser cutting software are you using? Both RDWorks & Lightburn automatically recognise internal cuts & do them first.

napter, 2018-12-03:
This feature would be helpful to me too. In order to cut holes first on a glowforge laser you have to import an SVG with the interior cuts (holes) a different color than the exterior cuts.

marmilicious, 2018-12-04:
@deece  We are using a Glowforge laser.   As @napter noted, the Glowforge needs to have interior lines be a different color in order to recognize and be able to cut those first.  But I could see this option being useful to others not using software as awesome as Lightburn.
(Lightburn really is great! Unfortunately we can't use it with the Glowforge.)

deece, 2018-12-04:
It's probably worth pinging Glowforge to add this feature then, it's obvious and essential, and should rely on third party applications to pre-process the data.

napter, 2018-12-04:
At the very least I think it would make sense for Deepnest to leave the original color of any path, currently it removes the colors and sets them all to black.

marmilicious, 2018-12-04:
@napter  I agree, it would be good if input has color that it be preserved.  In our case we are using Onshape to design parts.  Onshape exports parts as a single color though, thus the wish that Deepnest might have an option to color holes differently on export and streamline the process a bit.  Anything to help simplify the work flow is appreciated.

deece, 2018-12-04:
@marmilicious @napter Have you raised this with Glowforge? It's feature common in many other laser cutting packages (even Chinese ones), and with a few people asking, they might make it a priority.

napter, 2018-12-16:
@deece @marmilicious Is glowforge the only laser that allows you to change job settings based on color? A quick google shows the option to do color mapping with [Epilogs](https://www.epiloglaser.com/laser-machines/legend-laser/legend-color-mapping.htm). 

Regardless, if mapping cuts to colors isn't something to consider - keeping the original colors rather than losing them would be helpful. Thanks!

marmilicious, 2018-12-17:
@napter  Yes, Epilog has the ability to choose presets based on color.  Keeping original colors (as set in design software) could certainly be useful to those users.

deece, 2018-12-18:
@napter Most do, which is why the software should preserve colours. Any feature which deliberately alters the colours should be off by default.

Identification & sequencing of inner cuts is something that your CAM software is responsible for, so if it's missing, you should raise it there.

At the moment, it seems like @Jack000 has gone walkabout, so not only are there no patches to Deepnest coming in from him, but PRs are not getting merged either. If you want action, you'll need to look elsewhere.

zaped212, 2018-12-31:
I would love to have this feature as well. I have a full spectrum laser and it uses colors to set speed, power, cut order, and what should be raster engraved.

It would be great if each level of nesting could be given its own unique color so parts ( with holes ) nested inside other parts would have 4 different colors one for each interior line.

PodgPow, 2019-05-17:
One of the reasons that the holes are not recognized is set out in #59

rschenk, 2021-04-14:
I would love this feature as well—Deepnest preserves the original color of shapes when exporting as SVG. 

@deece you [mentioned](https://github.com/Jack000/Deepnest/issues/5#issuecomment-443371097) that some laser cam packages can automagically detect internal holes/features and sequence them first. That's fine if you only have internal holes that should be cut with the same settings as the outer perimeter, but supposing you wanted to also etch labels for those internal holes. Now you need a different power setting for the etch.  I set all this up with with different stroke colors in my design, but unfortunately Deepnest removes the colors and exports everything as black. It's a big pain in the neck to go back through and re-color the etches.

jestelle, 2021-04-14:
Me too! Unfortunately Jack isn't working on this project right now. But I'm interested in reviving it. Any developers out there want to help? See #123