Jack000/Deepnest #6: Support for part mirroring.

ExplosiveRunes, 2018-08-24:
Is there any plan to support mirroring of parts in addition to rotation?

marmilicious, 2018-08-24:
That would undoubtedly add some complexity to the nesting calculation, but indeed with certain part shapes that could allow better nesting.  I like the idea of this possibility.

RobsyRocket, 2018-09-02:
If this will be implemented, then please add an option to enable/disable this feature. This would avoid unwanted results if the material has different surfaces on each side.

kmostachov, 2020-06-18:
Mirroring ability checkbox per part would be a great addition to this awesome software.

Gingermittens, 2020-10-09:
An old feature request, but still absolutely relevant. With the ability to allow mirroring, this really would be absolutely class-leading when compared to fully-priced software, let alone donationware. Further to this, the ability to tag parts that can be mirrored and those that can't would be exceptionally useful.

luciancd, 2022-09-22:
I would also need part mirroring. A workaround could be adding several mirrored CAD parts by hand and start the calculation.