Jack000/Deepnest #7: DXF/SVG Conversion

lancer99, 2018-08-24:
Hello Jack

DXF/ SVG conversion login in this repo is referring to convert.deepnest.io URL. Will this be available and up all time or you have plans to open source that all also.

Thanks

Jack000, 2018-08-26:
the conversion just uses kabeja, I guess I can put up the code for that too

vupqtran, 2018-09-12:
Hi @Jack000 ,
Please put up the code for DXF/SVG conversion.

jaimecoj, 2020-09-01:
> the conversion just uses kabeja, I guess I can put up the code for that too

Did you upload the code? I would like to have the code to make some implementation

DayTimeDreamer, 2024-10-18:
I have implemented DXF -> SVG yet using kabeja, but kabeja doesn't look capable of doing reverse conversion.

Could you tell me how to implement the conversion from SVG to DXF? I would need to know to implement offline.

servant74, 2024-10-18:
Try https://github.com/Zigazou/svg2dxf