Jack000/Deepnest #23: Inconsistent path colors / stroke widths

jnabedian, 2018-09-25:
When exporting an SVG, some of the path widths are modified, and others have their colors modified/lost.  I created a simple test input file in Inkscape consisting of a square containing a few other rectangles (2 black, 1 red, 1 blue) and some blue text converted to paths (and applied "Break apart"). I had Deepnest pack 4 of these (but seems to be consistently bad no matter how many are added). The internal rectangles all lost their colors and became the same black as the outer square.  A few of the text paths turned black and had their stroke widths increased.  

Input:
[image]

Output (cropped): 
[image]

Source files:
[Square-Test.zip](https://github.com/Jack000/Deepnest/files/2415955/Square-Test.zip)

deece, 2018-09-26:
I've seen this too when exporting SVG, but colours export correctly in DXF (I do get some other artifacts like stray vectors which make DXF export unusable).

jnabedian, 2018-10-01:
Is this project still getting active updates?

dorkmo, 2018-10-01:
i think jack is taking a little time off