Jack000/Deepnest #31: Feature Request: Support Multiselect when importing files

phdoerfler, 2018-10-08:


mawildoer, 2018-10-20:
Came here to request the same.

schuylr, 2018-11-25:
We generate hundreds of custom individual SVG files on the fly when we get orders and assemble them into nesting tool in batches. This feature would be imperative for us.

laird, 2018-12-13:
I would love this. The core engine is awesome, but adding files one at a time via a GUI destroys productivity, and the point of automation is productivity. I would love to see this PR merged.

jsmwoolf, 2019-01-09:
Just want to let you guys know that the fix to this issue is now in a new development repository.  Since the owner has not responded for months, you should pick up the latest version from this [repository](https://github.com/InfernoEmbedded/Deepnest).

brend-3005, 2021-03-26:
Hi @jsmwoolf , when I run you code on my local host I just get this screen:

[image]

Did you get this also? Or do you know how I can fix this?
I know it is a long time ago :) but it would help me a lot