Jack000/Deepnest #34: C# Port

dharani811, 2018-10-16:
are there any c# ports for this available or do you have any plans to port this to C#? Please kindly let me know

dorkmo, 2018-10-17:
dont think there are any plans

petrasvestartas, 2018-10-20:
I am trying to work on C#. Currently I am using Clipper library to get NFP. In documentation it says that it is faster than using boost library.

I would like to ask if anyone of you could answer the question of mine in this thread:
https://github.com/Jack000/SVGnest/issues/75

In general I would like to know how nesting works. How you would nest a 3rd element (without talking about genetic algorithms)?

When you have multiple elements, do you need to find nfps in all of them, then boolean union them?
If this is right procedure, how you would place the element in multiple nps and be sure that there is no collision?

I am pointing at this picture and would like to know how blue line is found?
Is it cut by boundary sheet you are nesting ?

[image]

dharani811, 2018-10-22:
1. When you have multiple elements, do you need to find nfps in all of them, then boolean union them?
yes
2.If this is right procedure, how you would place the element and be sure that there is no collision?
Find the boolean union and then find Inner Fit Polygon,
Collision Free Range (CFR) = No Fit Polygon - Inner Fit Polygon.

 i.e CFR is that Blue line you're asking about. How and where do you place on the CFR is determined by heuristics you choose.  

3.If there would be any interest to collaborate I could create github repository based on clipper C# or native C++ whether using Clipper or Boost.

  I would love to but the author's involvement will be of great help.Nonetheless we can do it.

petrasvestartas, 2018-10-26:
Dear dharani811,

Thank you for the answers, I tried a simple nfp implementation:
https://vimeo.com/297354025

I got NFP by boolean union minkowski sums.
The was one issue, which I do not know if I did correctly, as I do not like my approach.

The problem is this:
When I get minkowski sums I get some holes, black outlines inside grey polygon

[image]

I removed them by looping each polygon and if there is a at least one point in the grey are, I remove the hole to get result like this:

[image]

The issue is that I am using Clipper to generate nfps and I check each polygon vertices if at least one of them is inside grey area and not lying on it, I remove full polygon. The problem is that clipper is a bit imprecise meaning some correction nfps outlines are by small increment inside grey area. What helps is to increase tolerance. 

**So the question I have is how you would remove minkowski difference generated holes?
Would you still use point in polygon inclusion check or some other method?**

dharani811, 2018-10-29:
Hello petrasvestartas ,

                                         The above polygon is Concave, a slightly complex one. Forming an NFP to Concave polygons needs a sophisticated approach. Of-course you can use Point-in-Polygon to remove islands in your case shown above but i am not sure about it's integrity. There are few other approaches to generate NFP for Concave polygons like "An improved method for calculating the no-ﬁtpolygon
Hamish T. Dean ,Yiliu Tu, John F. Raffensperger" which does deal with the cases like you mentioned above. Hope this helps .

petrasvestartas, 2018-10-30:
Do you know if there is a source code of this paper? :)

dharani811, 2018-10-30:
Hi Petras,

                 Sorry, i am not aware of any.

 Currently i am working on nesting with the following constraints:
1.Only Rectangular Sheets
2.No Holes in parts or sheets(i.e no part in part)
3.For NFP's with Holes , I am just letting them thru and yes they're
sometimes producing wrong results.But as of now the % is less .

           May be in future , in 30-40 days i will start implementing a
sophisticated NFP. Then i may help you.

Regards
Dharani

dharani811, 2018-10-30:
Hi Petras,

                 Sorry, i don't think there is for now.

 Currently i am working on nesting with the following constraints:
1.Only Rectangular Sheets
2.No Holes in parts or sheets(i.e no part in part)
3.For NFP's with Holes , I am just letting them thru and yes they're sometimes producing wrong results.But as of now the % is less .

           May be in future , in 30-40 days i will start implementing a sophisticated NFP. Then i may help you.

dharani811, 2018-10-30:
@petrasvestartas : If possible / permitted , you can share your code with me in mail. May be i can take a look and solve it for you.

Regards
Dharani

theveloped, 2018-11-08:
@petrasvestartas were you able to solve the problem of the extra holes that are generated when using the Minkowski sum? I currently do this by adding the shape itself (the grey part) to the computed Minkowski sum using a boolean union. This is however not applicable in all situations.

Also, did you and @dharani811 join forces in implementing a port? I currently have a simple port working implemented in Go. Let me know if you guys are interested in working together.

petrasvestartas, 2018-11-08:
Dear @theveloped ,

I just created empty repository here:
https://github.com/petrasvestartas/SharpNFP

I will upload my source files for generating multiple nfp I have.  But before that I would like to ask what software you are working with? My area of knowledge is within Rhino and Grasshopper.

Should we make console application but console application would not have any visual interface?

dharani811, 2018-11-08:
Hi guys , i have implemented a c# port based on https://github.com/exacloud/Nest4J, 

The o/p can be visualized via SVG. I have used the original clipper library for my operations. My library is pure C# and will not depend on anything else, i/p via XML files and o/p via SVG or console

I will upload the code within 3-4 days and then share the link here.

Let me know if you guys are interested in working together, let's discuss and looking very much forward.

theveloped, 2018-11-08:
@petrasvestartas we are currently working with multiple CAD packages and therefore are looking to indeed make a CLI version that simply takes polygons as input. Another option would be to make it available through a REST API like for instance: http://doc.powernestlib.com/

@dharani811 sounds good! I would definitely like to join forces on this.

petrasvestartas, 2018-11-09:
@dharani811

Sounding very interesting, I would like to test it too:)

dharani811, 2018-11-12:
@fel88 : Awsome man, I will check it out. looks nice though.

dharani811, 2018-11-16:
@fel88 ; @dorkmo ; @petrasvestartas ; @theveloped :  Here is my version of Nesting 

https://github.com/dharani811/NestingCShaprt