Jack000/Deepnest #39: Exports bad file with no "holes"

Dioxyk, 2018-11-08:
[image]
[image]

baulieuLCB, 2018-11-12:
It happened to me before, because the shapes I was importing were not closed. In the svg file, you need each shape to be a closed 'path' profile. (I guess that for now each curve is defined by one line / arc / bezier or whatever, they should be wrapped in a 'path' object)
I think you can do it with inkscape : https://graphicdesign.stackexchange.com/questions/100464/how-to-join-multiple-path-objects-into-a-combined-path-object-in-inkscape

Mowi, 2018-12-18:
it seems to me you should turn off the fill in inkscape and let it only show the stroke