Jack000/Deepnest #46: cannot call background.html

AttackerGry, 2019-01-07:
The application rebuilt with source code cannot call background.html. I don't know where the problem is.

AttackerGry, 2019-02-13:
I am sorry, this stems from my unfamiliarity with electron. Already successful

olegiv1, 2019-02-16:
I, too, not know electron - as correct - could you tell me please?

Jttwer, 2019-03-04:
Hello AttackerGry, I'm having the same problem. Could you tell me how you solved it? I'm also new to the world electron =)

codeplusplusplus, 2019-06-12:
Hi @AttackerGry , I have the same issue. Could you let us know how you solved the problem?

blackmad, 2019-08-04:
also getting an empty nesting screen, how did you debug/diagnose it?

jyc1000, 2019-10-23:
Hi @AttackerGry  I'm having the same issue and have tried various ways to build the addon with no avail. Could you please share just briefly how you resolved the issue? It would help immensely. Thanks.

odemakov, 2020-11-17:
I've spent 2 days to clarify this issue. 

The key located in the hidden background window. You should make it visible and look into developer tools error message. The messaged will be something like this:
```
Uncaught Error: The module '/Users/xxx/src/Deepnest/Deepnest/minkowski/Release/addon.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 50. This version of Node.js requires
NODE_MODULE_VERSION 57. Please try re-compiling or re-installing
```
The problem is that node is trying to use pre-compiled by author version of addon. Change the path to `dist/Release/addon.node` fixes the issue.

odemakov, 2020-11-17:
@Jttwer ,  @codeplusplusplus , @blackmad , @jyc1000 see comment above.