Jack000/Deepnest #49: All Parts broken to Independent Parts and Nested as Separate Parts .How to Group them as Single Part

JhonJebaraj, 2019-01-24:
[image]

The Parts image is broken as many small Images when Imported from DXF and SVG file . How to Make it as Single Image , We have attached the Image for your references,

myk3y, 2019-02-22:
You need to group the dxf before you export it.

thanhmeye, 2019-08-12:
I have the same issue as JhonJebaraj. I import several different parts of which some are imported in whole, some get broken into several small parts. This happened despite my making sure each part is grouped in the svg file. I have not tried with dxf though.

Uprising1, 2023-09-18:
Had the same problem, had to explode polylines in some drawings in order to import the file correctly.

bosd, 2023-09-21:
> Had the same problem

This is an unsupported version.
Please take a look at the community based fork.
Maybe the issue is fixed there.
https://github.com/deepnest-io/Deepnest