Jack000/Deepnest #50: dxf format

olegiv1, 2019-02-14:
DXF files are not read correctly - all information in DXF is not taken into account

deece, 2019-02-14:
Could you please be more specific about what is missed?

Also, make sure it's note a dupe of #10, #35 or #47

olegiv1, 2019-02-15:
Try the (attach - 2.dxf) file and see for yourself.Converter can not determine where the boundaries of the shape.Sometimes one wigura is placed on another.I'm trying to fix it, but I still can not build a project (I try in VS

olegiv1, 2019-02-15:
Try the (attach - 2.dxf) file and see for yourself.
Converter can not determine where the boundaries of the shape.
Sometimes one wigura is placed on another.
I'm trying to fix it, but I still can not build a project (I try in VS 2017)

deece, 2019-02-15:
I'm on a phone atm, but check to make sure your shape is a closed path.

olegiv1, 2019-02-15:
In the program - Grafis - this file is recognized normally
[2.zip](https://github.com/Jack000/Deepnest/files/2868098/2.zip)

olegiv1, 2019-02-18:
Why in debug mode, get the other sizes when you export data from DXF file?

closed 2019-05-16 by PodgPow