Jack000/Deepnest #51: Building project

olegiv1, 2019-02-15:
I have error (Win 10 x64, VS2017): 
D:\Deepnest-master\build\addon.vcxproj(51,398): error MSB4025: The project file could not be loaded.
Why in File addon.vcxproj strange characters at that position "C:oost_1_62_0"?

olegiv1, 2019-02-15:
if "C:\boost_1_62_0" -error
if "C:boost_1_62_0" -no this error

olegiv1, 2019-02-22:
I certainly started the project but had to use some "crutches". I'm used to uniqueness and certainty (I write in C++) of tools for program development. I will implement everything in C++. And Node js and Electron cause continuous bewilderment (((. 
Thanks Jack000!

deece, 2019-02-22:
Can you please share the build process? I'll include it into my fork where I'm merging all the Pull Requests

havid0707, 2019-03-08:
@PodgPow Can you share the build process? 

```
projects\deepnest\minkowski.h(3): fatal error C1083: Cannot open include file: 'boost/polygon/poly
gon.hpp': No such file or directory (compiling source file ..\addon.cc) [C:\Users\havid.zheng\projects\Deepnest\build\addon.vcxproj]
```
I have download boost and set the boost  folder to path env.  (
I am not familiar with C/C++)