Jack000/Deepnest #52: ratios are all over the place.

myk3y, 2019-02-22:
When importing a mm DXF file, from Sketchup or Fusion360 for example, the objects import as much larger than they are in the original drawing.

If I open the dxf file in Lighburn, the objects are exactly the right size.

[image]

If I import them into Deepnest, they are huge - those 1000+mm objects are actually 380mm

[image]

My settings are very straightforward

[image]

It looks like you have based your code on imperial units, instead of metric, which introduces all sorts of weirdness - if I change the SVG ratio to 2.54, as soon as I have done so, it changes to 2.5399999 or something similar.

Changing endpoint tolerance to 0,25mm ends up with it being something like 0.243899 

Surely everything should be in mm internally, with only the import and export in inches if you were unfortunate enough to not learn about the metric system at school? Performing ongoing inch/metric calculations introduces errors all the time, and when I am working on tolerances of 0.05mm, can get out of hand pretty quickly.

FranzBangar, 2019-03-04:
Hello,
I also have exactly the same problem - I have everything set to mm but no matter what my settings show, the import/export dimensions don't match.

deece, 2019-03-05:
You need to adjust the "SVG Scale" setting (under Import/Export). See https://community.glowforge.com/t/deepnest-resizing-svg-files/30736

myk3y, 2019-03-05:
> You need to adjust the "SVG Scale" setting (under Import/Export). See https://community.glowforge.com/t/deepnest-resizing-svg-files/30736

If you look at what I posted, I clearly put something in SVG scale.

It just doesn’t make sense. I only use mm,  why would I want to scale my SVG from anything other than 1:1?

I don’t use Inkscape or Illustrator for creating SVG, so the recommendations of 72 or 90(?) don’t apply.

If I put 2.54 in as the ratio, it changes after setting to 2.5399999 - clearly indicating a rounding error.

I am not the only one with this problem - it just isn’t a reliable app.

FranzBangar, 2019-03-05:
I ditched the scaling and all and added a circle of, say, 10 mm to the graphic. After nesting and export, I can scale the whole thing back so that the circle again has 10 mm.

jtme, 2019-03-12:
Same issue. When operating in mm the scaling is bit random. Changing the settings seems to have 0 impact. I was hoping to use canvas (or as referred in app - Sheet) as reference but it looks like output also crops/changes it. i.e. if i set 300x600 mm , it gets scaled down. But when i select it as sheet and save exported svg, it loses the original ratio/size and becomes the size of layout.

buikpijn, 2019-09-16:
Had the same issue; just set your units to mm to 10 and scaling is spot on.

myk3y, 2019-09-16:
You're right. I changed to 10mm and its working correctly.

Thanks. Great work.

amka-98, 2021-09-22:
May you please elaborate what you have done to get the exact dimensions?

buikpijn, 2021-09-22:
It’s currently all over the place again, it’s not working with the export feature of Fusion stable. One time it works, the other update it’s down the drain again. For that reason only i moved away from deepnest. It’s a struggle every time i use it not being size accurate.

yachal-upson, 2021-09-24:
Same issue (basically makes the software unusable/unreliable. PLEASE FIX THIS!) 

I tested by putting in a 100x100mm sample rectangle and found I got 283.5mm (hmm...). 
So ratio of 2.835 works as a workaround for me.

RikH90, 2026-09-24:
This issue is still ongoing as far as I can tell. Too bad :(