Jack000/Deepnest #54: Compiling Minkowski without Boost

samsonsite1, 2019-03-14:
Hello,

I would like to compile minkowski.cc without Boost.  I think all I need to do is replaced calculateNFP(), but I need a little guidance on how to do that.

Does the clipper library have the same functions?  If so, how would I replace calculateNFP() code with code from clipper library?  Please show an example, if possible.  Thanks.

PodgPow, 2019-04-04:
I think there is no reason to replace the boost in this project.

Jack000, 2019-04-05:
the only purpose of minkowski.cc is to integrate Boost, in order to use Boost Geometry. If you would like to remove it you can use the minkowski implementation in Clipper. This is what SVGnest does, so you can just use that code instead.

Boost geometry is a lot more reliable than clipper's minkowski implementation, which seems to produce small spurious paths which need to be filtered out.