Jack000/Deepnest #65: Problem with file export

yukiportes, 2019-08-09:
I put my svg files, the first time works great, but in the second time when I try to export, the program only export the file as a File, not like a svg file, I tried some times and I get the same result

snustav, 2020-11-14:
I had the same problem and it was driving me crazy, couldn´t find any answers on google so I tried everything. "Run as admin" didnt work, Uninstall and Reinstall didn´t work, restart the computer didn´t work. 
But at last i tried something that was too no brainer to work but it really did. 

So when you press "Export" and the file locations window shows up, you need to write your file name and end it with the same file you chose to export. 

So for example: i want to export dxf, so i press export dxf and then name it to "thisisatest" and the ends it with **.dxf**
it looks like this: thisisatest.DXF

Erwin1399, 2021-01-21:
Thank you man, 
I was close to trowing out the pc. 
But now I can export to dxf again :)

zigm4Rift, 2023-11-03:
hi there despite adding the file type it still hangs. exporting both svg and dxf of 152 shapes.

bosd, 2023-11-03:
This repo is not actively maintained anymore.
There is a community based fork at: https://github.com/deepnest-io/Deepnest