Jack000/Deepnest #68: Nesting does not use multiple sheets.

Danielaaroe, 2019-09-01:
It is indicated by the sheet-counter that nesting should be able to nest between multiple sheets. When my first sheet is full it does not create a sheet number 2, but keeps working without finishing a single version. Sheet is 2400mm*1200mm(full sheet) and components are up to 800mm diameter. GA population set to 5 to speed up first version of nest, but still nothing.

ecridlebaugh, 2019-10-03:
Having this same issue

vanThijs, 2019-11-11:
Deepnest doesn't do this automatically, but you can easily add another sheet: Mark you sheet as the sheet and increase the quantity to 2 or more. In my experience, deepnest will try to minimise the sheet use.

Danielaaroe, 2019-11-26:
Unbelievable. I feel like an idiot. This issue is now solved. Thanks @vanThijs :)

closed 2019-11-26 by Danielaaroe