Jack000/Deepnest #76: Setting Deepnest

rubenpezu, 2019-12-05:
Hi all,
I'm new with Deepnest. I'm trying to create some nesting but I have always two problems:

1. I can't import dxf. Always appear "could not contact file conversion server".

2. When I import .svg appear with different dimension than in .dxf

Thanks a lot for your help.

andres-cm90, 2019-12-05:
on the settings make sure that they match the units with which you created the .dxf file. I would try it with a square part, choose any dimensions and after moving the settings, once you import it, make sure the size matches the real size.

andres-cm90, 2019-12-05:
For the importing, I was having that problem. Try switching the format to R12/LT dxf file type instead of a more recent one.

vmario89, 2019-12-05:
mhh some annoying workaround to solve dxf handling would be to use InkScape or sk1 for example to make up another import format.