Jack000/Deepnest #78: Multiple nest errors (DXF R12/LT)

andres-cm90, 2019-12-05:
Having some problems nesting some geometries. I made sure they're all polylines and closed contours, with no hidden layers nor lines, however, 3 issues can be observed in the pictures:
1. Overlapped geometry
2. Geometry outside the sheet edge
3. Geometry with extra line overlapping a part

[image]
[image]
[image]

marianoju, 2019-12-14:
I have observed overlapped geometry as well, see screenshot. 
[image]

Even though I have added more than 2 sheets, Deepnest tries to fit in all objects on 2 sheets. Only work-around I have found is closing Deepnest, and starting the process from scratch, that is: importing all DXF files and start nesting. This error occurs only sporadically.

marianoju, 2019-12-18:
Overlapping appears to be less likely when imported objects have previously been joined to polylines.

Enggad, 2021-06-22:
I can not to import any item

Enggad, 2021-06-22:
[image]

Enggad, 2021-06-22:
Any body can help me to solve this problem

servant74, 2021-06-22:
> <> ... Jack

I just downloaded deepnest for windows, and imported a sample svg
sofar no issues.

Could you explain what you are doing, possibly share the svg you are using
for testing?

marianoju, 2021-06-22:
@Enggad Your issue does not appear to be connected to multiple nest errors (see above), so I suggest you create [a new issue](https://github.com/Jack000/Deepnest/issues/new/choose).