Jack000/Deepnest #82: [Feature Request] - Default Sheet

sgtpepperaut, 2019-12-16:
Most of us here always work with the same Machine/Bed Dimenions. It would be great if a default sheet size could be included right from the start. Its not much but it would help with the workflow. Also I would think the program should auto determine the number of sheets needed. It could start by assuming squares around all objects and add these squares? this should give the maximum area needed. Then from the sheet dimension the program would have a starting point to work its way down? The use case here is that i have many parts and i dont even know how many sheets i need so i enter a stupidly large amount which might actually hurt the calculations?

oberon227, 2020-02-22:
@sgtpepperaut: This doesn't help you with the default sheet, but I _can_ help you with the "auto determine the number of sheets".  Once you've imported all your SVG objects, add your rectangle to act as your sheet, set the quantity of those rectangles to some large number, and check the Sheet checkbox.  Then when Deepnest is nesting, it'll tell you "X sheets used".
Seems like this is a feature, but not a very well documented one.

iliis, 2023-08-27:
There's also a 'plus' button to add a rectangle on the bottom left:
[image]