Jack000/Deepnest #83: Not working - how to debug?

raveren, 2019-12-18:


raveren, 2019-12-18:
Oh, I figured it out, there's no tutorial of any kind though.

niceday11, 2020-07-21:
Hi @raveren 
I have encountered the same problem. Please teach me how to solve it. Thank you very much

niceday11, 2020-07-21:
Hi @raveren 
I have encountered the same problem. Can you teach me how to solve it? 
Thank you very much.

raveren, 2020-07-26:
Aw man sorry, this thing never worked for me properly, I mean I did make it work on my machine, it was a sequence of user actions thing that you have to do in correct order for it to function, but I don't remember now, and once it failed to optimize any of my real world files I never used it again.

niceday11, 2020-07-27:
I am very grateful to receive your reply, but unfortunately you forgot some details. I need to use it to do some debugging and then proceed to the next step. Thank you still

rusticspartan, 2022-12-13:
I realize this is an old post but figured out how to use the program and wanted to share. First, import an svg/dxf file using the top button. Then, you need to select the size of the sheet you want the items to be placed on. At the bottom, you will see a square button with a plus sign in the middle, click that and choose your size. Next, chose how many copies of each vector(s) you want. Click start nest. It will keep moving items on its own til you tell it to stop. When happy, click export and choose a file name (make sure to include .svg at the end).