Jack000/Deepnest #84: no rotate or degree step rotate feature

cworf, 2020-01-24:
this may not be a popular request but i figured id put it down anyways. I am using this to organize wooden laser cut stickers on a sheet of wood veneer. it would be cool to have the ability to specify that only 0° and 180° rotations may be tried, this is due to the fact that wood veneer grain only flexes in one direction so the cut-out's must be oriented properly with the grain of the wood in order to be able to flex in the right direction to be stuck right side up to a curved surface such as a water bottle.
GREAT WORK on this.

balunist, 2020-01-24:
Currently you have the default SVG scale settings at 72 units/inch.   To me it would make more sense that this would be the modern standard of 96 units/inch.   Since this can be set by the user it is not a big deal but it might present less confusion.   It would also match the SVG scale exported by a Fusion 360 add-in MapBoards Pro.

cworf, 2020-01-24:
@balunist I think you've commented on the wrong post.

floppy247, 2020-06-19:
cworf, for me your request is very popular. This tool is all I looked for a real long time, BUT we're working with all kind of textiles, some of them with block stripes and structures. If the tool turns them the tool is pretty useless tu us.
I hope theres a way to develop the "no-rotation"-option 👍

HangliderUK, 2020-12-05:
> cworf, for me your request is very popular. This tool is all I looked for a real long time, BUT we're working with all kind of textiles, some of them with block stripes and structures. If the tool turns them the tool is pretty useless tu us.
> I hope theres a way to develop the "no-rotation"-option

Hi @cworf  and @floppy247 , I downloaded v1.0.5 a few days ago and also wanted to limit rotations.  I was able to do this by going to settings and setting "Part rotations" to 1, which me at least, resulted in no rotations...so might work for the textile pattern problem.
In addition I found that setting the value to 2 gave me 0 and 180 degree rotations which I think should help with the "wood grain" issue?

Hope this helps!