Jack000/Deepnest #87: Harfbuzz version too old (1.2.7)

tiket18, 2020-02-04:
./Deepnest-1.0.5-x86_64.AppImage 

(deepnest:41438): Pango-ERROR **: 11:10:21.253: Harfbuzz version too old (1.2.7)

Trace trap/stop (memory stack reset to disk)

uname -r
5.4.15-arch1-1

Doc91b, 2020-04-24:
I'm getting the same result. 

```
$ ./Deepnest-1.0.5-x86_64.AppImage 

(deepnest:4080): Pango-ERROR **: 17:31:06.473: Harfbuzz version too old (1.2.7)

Trace/breakpoint trap (core dumped)

```
```
$ uname -r
5.4.34-1-MANJARO
```

Pango:         1:1.44.7+11+g73b46b04-1
Harfbuzz:     2.6.5-1
Electron:      8.2.3-1

Dug around a little and found the following which I hope will be useful:
[https://github.com/electron/electron/issues/20348]

ThatRusty, 2020-05-18:
I'm getting this same error

concifederico, 2020-06-29:
Same error
`$ uname -r
5.4.0-39-generic`

concifederico, 2020-07-05:
Solved! (Ubuntu 20.04 ) You have to downgrade libpango LOCALLY to 1.42.4-7
This is done by extracting the .so. files from the .zips files in the same folder where Deepnest-1.0.5-x86_64.AppImage (https://deepnest.io/) is. Here the links:

https://packages.ubuntu.com/eoan/amd64/libpango-1.0-0/download
https://packages.ubuntu.com/eoan/amd64/libpangoft2-1.0-0/download
https://packages.ubuntu.com/eoan/amd64/libpangocairo-1.0-0/download

Regards!

Federico

closed 2020-07-06 by tiket18

nicholas-gh, 2020-08-24:
Maybe this issue should be (re)opened until the AppImage works on Ubuntu 20.04 as-is?

I ran in to the problem and thought to come here to search for answers - great there was one, thank you! But probably the package can be fixed?

keymon, 2021-01-03:
Yeah, the issue should be reopen.

Also, unfortunately the links provided by @concifederico are broken :(

keymon, 2021-01-03:
Alternative solution here: https://github.com/Jack000/Deepnest/issues/109#issuecomment-710869498

propcoder, 2022-09-11:
> Alternative solution here: [#109 (comment)](https://github.com/Jack000/Deepnest/issues/109#issuecomment-710869498)
How exactly should we use the ldd with AppImage?
This is what I get:
`$ ldd Deepnest-1.0.5-x86_64.AppImage | grep pango`
`        not a dynamic executable`