Jack000/Deepnest #98: parts overlap

bladestonez, 2020-05-10:
every time i use this parts are placed on completely on top of each other or at the best overlapped. happens 100% of the time.
[image]

mza1979m, 2020-12-31:
I'm also having the same problem now. I've used the program many times without issue in the past. Now I'm experiencing this with a specific svg file.

ClockworkBastard, 2024-07-03:
This issue is really freaking me out :((( The program do know that parts did overlaped drawing a crosshatch and glowing with a beautiful blue color... Looks like it is work in progress to be fixed. We'll have to wait
[image]

bosd, 2024-07-03:
> This issue is really freaking me out :(((

Can you check in the community based fork if the problem still exists?

bosd, 2024-07-04:
Deepnest is abondoned by the original author.
There is a community based fork at this repo: https://github.com/deepnest-io/Deepnest

ClockworkBastard, 2024-07-04:
> Deepnest is abondoned by the original author. There is a community based fork at this repo: https://github.com/deepnest-io/Deepnest

Well, I did used Deepnest.io , downloaded from the provided web page on the project's one. **YET there is still version 1.0.5**
So I'll check out the one from github as latest - 1.2.0

bosd, 2024-07-04:
Thanks for checking.
The new repo name is confusing.

(It is  it related to deepnest.io)
sorry should have picked something else when creating it. 
But now I no longer have the super powers to change it.

Can you please create a ticket there.
We occasionally have a dev working on this thing over there.

ClockworkBastard, 2024-07-05:
> (It is not related to deepnest.io)
> sorry should have picked something else when creating it.
> But now I no longer have the super powers to change it.

Well, if you are the creator then you do have powers to rename it! Or else, ask the one who is administrator there or smth
https://docs.github.com/en/repositories/creating-and-managing-repositories/renaming-a-repository