Jack000/Deepnest #99: Keep colors

3fonov, 2020-05-19:
I've seen somewhere, that deepnest keeping colors. But cannot reproduce: every file i import become black and white. Is there any way to keep colors after import? It would be great for engraving and cutting after that.

joebywan, 2020-11-11:
Did you manage to figure this out?  I've just encountered this too.

MV10, 2020-12-11:
Way back in 2017, Jack replied to this review to say that he added color support. Colors/layers aren't preserved for me, either. Unfortunate, it looks as if he may have abandoned the project.

https://www.ponoko.com/blog/digital-manufacturing/definitive-guide-nesting-software-laser-cut-designs/

rschenk, 2021-04-14:
Having done a decent amount of poking around with this today on 1.0.5, this problem only seems to rear its head when Merge Common Lines is set to true and a merge occurs. _If the merge common lines feature is turned off, the original colors are maintained._

Now I can see how if you're merging two lines of different colors, you're kinda hosed because there is no original color to keep. But if merging two lines of the same color, it would be nice if the original color of the two lines that were merged was maintained.

philspitlerSF, 2022-01-15:
@rschenk I tried turning off "merge common lines" and it is still not retaining the colors... Everything is coming back in with a solid black fill. 

Any ideas?

UnfoldAntwerp, 2022-05-02:
This baffles me too, I thought it just didn't work until few days ago, when it suddenly worked on one file. But not on recent files anymore. So the must be something I did that made it work once. Maybe a specific SVG setting?

caseprince, 2023-08-04:
I've figured out 2 more causes for this issue and workarounds for each:
1. Input SVG's `<style>` tags are not saved in output SVG. If your colors are coming from CSS declarations instead of inline `stroke` or `fill` attributes they won't be preserved. The workaround is to manually copy `<style>` tags from the input SVG, into the output SVG, by opening the files in a text editor like notepad or VSC.
2. Some (but not all) `<polygon>`s were being stripped of their class assignments . (I _think_ these may have started as `<rect>`s that were converted into  `<polygon>`s when rotated but not 100% sure.) All my classless polygons happened to want the same class, so I was able to find and replace `<polygon points` with `<polygon class="st1" points`.

Both seem like they would be relatively simple fixes, but until we get a PR or another fork the workarounds should hopefully help!