Jack000/Deepnest #102: nested into each other

Barthiboy, 2020-06-29:
Hi,
i have a small problem.
Deepnest ist nesting some svg into each other.
[image]
[image]

I tried different options. I have groped the lines and so on. I tried to minimize the xml code. But nothing 
[Eule4Ast.zip](https://github.com/Jack000/Deepnest/files/4844564/Eule4Ast.zip)
[SchneckeBesprechung.zip](https://github.com/Jack000/Deepnest/files/4844567/SchneckeBesprechung.zip)
Does anyone know this issue?
How can i fix this problem? It seems to be a problem in the svg file.

concifederico, 2020-06-29:
I think its a problem with the SVG, go Inkscape and paint it in colors, then work with the Union and Difference command until all parts you want are colored.

Barthiboy, 2020-06-29:
Thank you for your fast response. I tried the union and difference, but the problem is, that i want to engrave the inner lines and to cut just the outline. When i do the union and difference the properties of the outlines are the same as the inner once. So the cutting line is green an the engraving lines are black. Apart from this i tried to import the new file with the same result.

jestelle, 2021-04-09:
I'm running into this problem with a much simpler shape:

[image]

I want to cut the black line and just score the red lines.

Deepnest gets confused and can't really deal with this well.

Is there any way to fix it? Deepnest always assumes my scores are cuts, and tries to nest smaller pieces inside:
[image]