Jack000/Deepnest #107: SVG->DXF Conversion

jaimecoj, 2020-09-02:
As commented in other issue, DXF/ SVG conversion is referring to convert.deepnest.io URL. 

We need offline support and I have implemented DXF -> SVG yet using kabeja thanks to author comment, but kabeja doesn't look capable of doing reverse conversion. 

@Jack000 , could you tell me how is implmented conversion from SVG to DXF? I would need to know to implement offline.

Regards

tketel, 2021-01-29:
I am also in the same boat. I have many DXF files that I need to nest, but I have to do it on an offline system, because the files are confidential.

9swampy, 2021-06-19:
Ditto on the DXF->SVG; to import I've remoted to the Kabeja CLI but needing to go the other way.

Did anyone make progress here? 

I've found https://github.com/codelibs/libdxfrw which could maybe be used similarly to the C++ Minkowski but way out of my comfort zone.

DayTimeDreamer, 2024-10-18:
I have implemented DXF -> SVG yet using kabeja, but kabeja doesn't look capable of doing reverse conversion.

Could you tell me how to implement the conversion from SVG to DXF? I would need to know to implement offline.

jaimecoj, 2024-10-19:
> I have implemented DXF -> SVG yet using kabeja, but kabeja doesn't look capable of doing reverse conversion.
> 
> Could you tell me how to implement the conversion from SVG to DXF? I would need to know to implement offline.

Hello, I managed to convert SVG->DXF by using inkscape script "dxf_outlines", you have more info of usage here https://graphicdesign.stackexchange.com/questions/150605/how-to-export-an-svg-file-to-dxf-r14-using-inkscape-command-line