Jack000/Deepnest #109: Harfbuzz version too old (1.2.7

gregers666, 2020-09-16:
(deepnest:65177): Pango-ERROR **: 08:06:01.108: Harfbuzz version too old (1.2.7)

Distributor ID:	Linuxmint
Description:	Linux Mint 20
Release:	20
Codename:	ulyana
Linux lucyfer 5.4.0-47-generic #51-Ubuntu SMP Fri Sep 4 19:50:52 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Package: libpango-1.0-0
Architecture: amd64
Version: 1.44.7-2ubuntu4
Multi-Arch: same
Priority: optional
Section: libs
Source: pango1.0
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian GNOME Maintainers <pkg-gnome-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 418
Provides: libpango1.0-0 (= 1.44.7-2ubuntu4)
Depends: fontconfig (>= 2.1.91), libc6 (>= 2.14), libfribidi0 (>= 1.0.0), libglib2.0-0 (>= 2.59.2), libharfbuzz0b (>= 2.1.1), libthai0 (>= 0.1.25)
Breaks: libpangocairo-1.0-0 (<< 1.44.7), libpangoft2-1.0-0 (<< 1.44.7), libpangoxft-1.0-0 (<< 1.44.7)
Filename: pool/main/p/pango1.0/libpango-1.0-0_1.44.7-2ubuntu4_amd64.deb
Size: 162008
MD5sum: 6c8794e564f21e7af61be6206ef4d803
SHA1: 6a8cfc106ba18607bfb00d6eb33af51dc8e09f3c
SHA256: 5ab27de09a2f0dbd0a4959c7427cfa512ca1e01549fe0767b6e0db8d00689f3b
Homepage: https://www.pango.org/

dravac, 2020-10-17:
Hi @gregers666 

You have to downgrade Pango to version 1.42.4.
Here is a solution with the compilation of the library.

```
wget https://deepnest.io/Deepnest-1.0.5-linux.zip
unzip Deepnest-1.0.5-linux.zip

mkdir pango
cd pango
wget ftp.gnome.org/pub/GNOME/sources/pango/1.42/pango-1.42.4.tar.xz
tar -xf pango-1.42.4.tar.xz
cd pango-1.42.4
./configure
make
cd pango/.libs/

cp libpango-1.0.so* ../../../../Deepnest-1.0.5-linux/.
cp libpangocairo-1.0.so* ../../../../Deepnest-1.0.5-linux/.
cp libpangoft2-1.0.so* ../../../../Deepnest-1.0.5-linux/.
cp libpangoxft-1.0.so* ../../../../Deepnest-1.0.5-linux/.
```

gregers666, 2020-10-23:
Hello!

Thank you it works really nice!

Best regards

closed 2020-10-23 by gregers666