Jack000/Deepnest #111: Words breaking into letters

urban2015, 2020-09-21:
HI,

I want to use this with words but when i attached the letters together the program breaks the words into letters. I have tried using illustrator and inkscape both failed.

Could you help?

amunizp, 2020-09-22:
Using inkscape maybe this might help: make each word to paths then merge them using the union tool.

urban2015, 2020-09-22:
I have tried union the letters together and still the letters behave as separate ones. If i import the same svg file into another program the letters are not separate so the program is separating them.

amunizp, 2020-09-23:
Can you overlap the letters just a tiny bit? After the nesting is done you can then return it to the original letter space.

urban2015, 2020-09-24:
Yeah but its not so easy to get the letters back after... I tried connecting letters with shapes to make the letters all joined up but after the nest it was impossible to remove the shapes.

amunizp, 2020-09-25:
Do you mean wrapping the letters with a line? Using outset? Because that was my next suggestion.