Jack000/Deepnest #113: Manual nesting mode improvement?

erosenst, 2020-12-16:
Is it posible to add manual nesting mode to move/rotate parts then continue automatic mode? It make fair good nestings but manual mode corrections is usually need to get best results.

I have cnc plasma cutter. Is it posible to add lead in and lead out curves to parts? My plasma needs some hundred milliseconds to start and stop so lead in and out curves is neccesary. Cutter compensation ~1mm?

Integration to dxf2gcode to produce g-code output?

Thank you

Durss, 2020-12-16:
No doubt these features would be awesome to have !

But they're are far from trivial to implement and the owner of the project hasn't been active for a while now.
Don't expect them to be added anytime soon unless someone very nice (and smart) decides to spend time on it ^^

ulfaslak, 2021-01-30:
Yes this is a pretty big feature to implement. And remember this is Open Source. Sharing code (especially of this high a quality) is an act of compassion, and as a community we have a responsibility to contribute. So if you want a new feature, think about having a go at implementing it yourself and making a pull request. There's nothing more gratifying than seeing other people adopt and build on the code you shared with them.

9swampy, 2021-09-03:
If you're on Windows I've taken the C# port and added manual editing (among other things I needed) [https://github.com/9swampy/DeepNestSharp](https://github.com/9swampy/DeepNestSharp)