Jack000/Deepnest #116: SVGs overlapping

l3kris, 2021-01-07:
My shapes keep overlapping:

[image]
[image]

herault-guillaume, 2021-01-16:
same issue with Deepnest 1.0.5

[image]
[image]

9swampy, 2021-09-03:
I was having similar issues and if you're on Windows I've taken the C# port and (tbc) fixed for my parts (among other things I needed) [https://github.com/9swampy/DeepNestSharp](https://github.com/9swampy/DeepNestSharp).

One tip is make 100% sure your part is perfectly closed. Even the tiniest gap on either an outer or inner polygon leaving it open would break the code.

If you're not on Windows but your overlaps are consistent I'd still be interested in part file if you'd be happy to forward it.

marmilicious, 2021-09-03:
> One tip is make 100% sure your part is perfectly closed. Even the tiniest gap on either an outer or inner polygon leaving it open would break the code.

Ah thank you for this tip! Maybe there could be a check for this, highlighting the problem shape, and possibly even an option to fix/close gaps that are smaller then x amount?

9swampy, 2021-09-03:
No worries; it isn't the only reason for the overlapping to occur though. Just one you can check for and fix yourself without having to get knee deep in code.