Jack000/Deepnest #117: Approximation of curves

ajlapp, 2021-02-03:
I'm using Deepnest all the time with my waterjet cutter!  It's amazingly handy.

That said the approximation of curved surfaces seems very poor and no setting I've found seems to be satisfactory.

The curves look great in the nesting preview but the DXF exports have very course linear approximations of the fine curves.  Is there anyway to improve this?

[image]
[image]
[image]

andrewwong2000, 2021-03-19:
I noticed this too when trying out Deepnest.. if it preserved arcs when nesting a DXF it would be usable.
Due to this issue (affects cut quality and performance)  and the occasional output overlapping parts I just do manual nesting
in Sheetcam.

Will keep an eye on this and see if it changes.

PaulShroom, 2024-06-22:
Would it be hard to relocate the original using a script after the nesting has finished?

PaulShroom, 2024-06-22:
"If it preserved arcs when nesting a DXF it would be usable." Yes, it would be a game-changer.

bosd, 2024-06-23:
Hi, Thanks for your interest in deepnest.
Deepnest is abondoned by the original author.
There is a community based fork at this repo: https://github.com/deepnest-io/Deepnest
Maybe this issue is fixed there. Please check.