Jack000/Deepnest #121: Export Sheets in DXF File

Zaaphod, 2021-03-25:
It would be extremely helpful to get the sheets exported in the DXF File, Or at least the 0,0 point of each sheet, Especially when Space between parts, or multiple sheets are used as it becomes difficult to know where to set the 0,0 point for the CAM operations.

brend-3005, 2021-03-26:
Hi @Zaaphod did you get the code working? when I just drop the folder on my local host I get this screen:

[image]

TrianglemanTriangleman, 2025-02-19:
I agree with this request.  If we were given the option to include the sheet in the export that would be extremely helpful to visualize remaining stock in CAD environments and documentation.

bosd, 2025-03-25:
@TrianglemanTriangleman Can you open an issue with this request over at the new fork? https://github.com/deepnest-next/deepnest.