Jack000/Deepnest #123: Revive this project?

jestelle, 2021-04-14:
Hi all,
It seems there's lots of interest here on Deepnest, but Jack doesn't have time to dedicate to it.
I'm interested in potentially giving it a bit of love... any other software developers out there willing to chip in with me?
Express some interest on this thread and we'll see if we can get this thing moving again :D 

Cheers!

dorkmo, 2021-04-14:
there have been a few people that have forked it over the years that attempted to move forward. really like this project and hope theres people interested in developing on it!

jestelle, 2021-04-14:
Any of those forks active?
I can see a few with a number of commits, but most haven't been updated in a while.

dorkmo, 2021-04-14:
unfortunately i dont think so. but it might be worth reaching out to those guys

rschenk, 2021-04-14:
I'm tempted to take a crack at #99. I've never built an electron app, so I'm not sure where to start with that, but am solid frenemies with Javascript so I think I could give a decent stab at it.

jestelle, 2021-04-14:
It seems the first real challenge is just getting Deepnest building :D and
maybe documenting how to do that.

Jack mentioned some API changes in some of the dependencies, so the first
challenge would be getting a fresh build working.

After that, some of these bugs and creature requests don't seem like they'd
be too challenging to address.

kelyamany, 2021-04-16:
I wasn't able to build the application, at least the native C++ library. Would love to participate especially with handling the DXF files if someone was able to build it successfully and document the steps in details.

dorkmo, 2021-04-16:
Take a look in issue #51

kelyamany, 2021-04-19:
> Take a look in issue #51 , #46 , #34 , #7 , #4 and #2

Great, thanks a lot. I managed to compile it on Windows 10 after a lot of trials, here are the steps if I remember correctly:
1. Download and unpack [Boost 1.62](https://www.boost.org/users/history/version_1_62_0.html) and unpack it to C:/boost
2. `npm install --target=1.4.8 --arch=x64 --dist-url=https://atom.io/download/atom-shell"`
3. Run `.\node_modules\.bin\electron-rebuild.cmd`
4. Copy the contents of build/Release to minkowski/Release
5. `npm run start` or `electron .`

I also have Visual Studio 2019 and Python 3 installed in my environment, not sure if they make a difference with anything

9swampy, 2021-06-17:
Thx @kelyamany those steps really helped; but for others trying same I can add more info

1. Download Boost 1.62 and unpack it to C:/boost (note this must be same drive as the main code you're building; my code was on D so I needed to put at D:/boost)
2. You need Python 2.7 (I got 2.7.18 because gyp(?) isn't compatible with Python 3)
3. You need Visual Studio (or possibly the SDK; as @kelyamany I had VS on the box already)
4. `npm install --target=1.4.3 --arch=x64 --dist-url=https://atom.io/download/atom-shell"` (at least I couldn't find any 1.4.8)
5. Run `.\node_modules\.bin\electron-rebuild.cmd`
6. Copy the contents of build/Release to minkowski/Release
7. `npm run start or electron .`

That got me to a running application that executes the nest happily; unfortunately I have one bump - the preview graphic works fine but the view of the resultant nest doesn't show anything and it seems like the exported file isn't valid. Close but no cigar. Anyone got any advice to offer?

On another note I also came across https://github.com/fel88/DeepNestPort which is a more actively supported port in C#.

servant74, 2021-06-17:
I hate to suggest this, because I know I don't have the know how.
Could someone please make a 'deepnest for rookies' youtube video.  Install
and get it going.
Other than being a 'handy' reference for reference, it might get some new
blood into helping
or at least keeping the project 'alive'.

I get all kinds of 'good ideas', but run out of steam before they are
implemented.

Babuh, 2021-07-06:
Me and my team are very interested about this project as well. We have check the c# port and it is great to see that the interest is growing again ! 
However we are more interested in SVGNest as a web application is way more flexible and easier to deploy. Do you guys have a particular reason to use the c# port instead of SVGNest ? Do you know a more active fork or community around SVGNest as the main repo seems abandoned ?

Please let me know if you are interested in reviving SVGNest as well !

9swampy, 2021-07-07:
DeepNest sends the files to a remote server for conversion, that's a big limitation. I'd expect SvgNest would do the same but havn't looked. I did some work on DeepNest and got a fix for import; but after the nest you still need to send it for conversion to do the export so no real improvement.

wangjian2020, 2021-09-19:
How can I contact the writer? Sent him an email at jak000@gmail.com but didn't get any response... I need a new feature and I am willing to donate~

Pinaute, 2022-03-09:
Hi,
For those interested, I managed to build a docker for deepnest. 
Based on the improvement work of @9swampy

dockerfile
```
FROM debian:bullseye-20220228-slim

RUN apt update && apt -y install \
  build-essential \
  ca-certificates \
  tar \
  wget \
  git \
  --no-install-recommends \
  && rm -rf /var/lib/apt/lists/*
  
RUN git clone --depth 1 --branch develop https://github.com/9swampy/Deepnest.git /opt/deepnest

RUN wget "https://sourceforge.net/projects/boost/files/boost/1.76.0/boost_1_76_0.tar.gz/download" -O /tmp/boost_1_76_0.tar.gz \
  && mkdir /opt/deepnest/boost_1_76_0 \
  && tar -xzvf /tmp/boost_1_76_0.tar.gz -C /opt/deepnest/boost_1_76_0 --strip-components 1 \
  && rm -rf /tmp/boost_1_76_0.tar.gz \
  && cd /opt/deepnest/boost_1_76_0 \
  && ./bootstrap.sh --prefix=/usr/ \
  && ./b2

COPY binding.gyp /opt/deepnest/binding.gyp

# Python is required by nvm
RUN apt update && apt -y install \
  python \
  --no-install-recommends \
  && rm -rf /var/lib/apt/lists/*

# https://stackoverflow.com/questions/25899912/how-to-install-nvm-in-docker
SHELL ["/bin/bash", "--login", "-c"]

RUN cd /opt/deepnest \
  && wget -qO- https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.1/install.sh | bash \
  && chmod +x /root/.nvm/nvm.sh \
  && . /root/.nvm/nvm.sh \
  && nvm install 12 \
  && nvm use 12 \
  && npm install --arch=x64 \
  && ./node_modules/.bin/electron-rebuild

RUN rm -r /opt/deepnest/minkowski/Release \
  && cp -a /opt/deepnest/build/Release /opt/deepnest/minkowski/Release

# Required by electron, I use ldd to print library dependencies 
RUN apt update && apt -y install \
    libglib2.0-0 \
    libnss3 \
    libnspr4 \
    libgdk-pixbuf-2.0-0 \
    libgtk-3-0 \
    libpangocairo-1.0-0 \
    libpango-1.0-0 \
    libatk1.0-0 \
    libcairo2 \
    libdbus-1-3 \
    libxext6 \
    libx11-6 \
    libxcomposite1 \
    libxrender1 \
    libx11-xcb1 \
    libxcb1 \
    libxcursor1 \
    libxdamage1 \
    libxfixes3 \
    libxi6 \
    libxtst6 \
    libxrandr2 \
    libxss1 \
    libasound2 \
    libatk-bridge2.0-0 \
    libatspi2.0-0 \
    libcups2 \
    --no-install-recommends \
    && rm -rf /var/lib/apt/lists/*
    
# [12124:1012/192744.263024:FATAL:setuid_sandbox_host.cc(157)] The SUID sandbox helper binary was found, but is not configured correctly. Rather than run without sandboxing I'm aborting now. You need to make sure that  /opt/deepnest/node_modules/electron/dist/chrome-sandbox is owned by root and has mode 4755.
RUN chown root /opt/deepnest/node_modules/electron/dist/chrome-sandbox
RUN chmod 4755 /opt/deepnest/node_modules/electron/dist/chrome-sandbox

COPY Settings /root/.config/Deepnest.io/Settings

RUN mkdir /root/workdir
VOLUME /root/workdir

WORKDIR /opt/deepnest
ENTRYPOINT [ "/opt/deepnest/node_modules/electron/dist/electron", ".", "--no-sandbox" ]
```
You need these two files below in the same directory as the dockerfile

binding.gyp
```
{
  "targets": [
    {
      "target_name": "addon",
      "sources": [ "addon.cc", "minkowski.cc" ],
      'cflags!': [ '-fno-exceptions', "-m64" ],
      'cflags_cc!': [ '-fno-exceptions', '-fPIC -m64' ],
      "include_dirs" : ["<!(node -e \"require('nan')\")", "/opt/deepnest/boost_1_76_0/"]
    }
  ],
}
```

Settings
```
{"units":"mm","scale":"96","spacing":12.09448818897638,"curveTolerance":0.72,"rotations":4,"threads":"8","populationSize":10,"mutationRate":10,"placementType":"box","mergeLines":false,"timeRatio":0.5,"simplify":false,"dxfImportScale":"1","dxfExportScale":"72","endpointTolerance":0.36,"conversionServer":"http://convert.deepnest.io","useConverterDeepnestIo":false,"kabejaPath":"D:/kabeja-0.4"}
```

1. build
`docker build -t deepnest:debian-bullseye .`
2. run
`docker run --rm --name deepnest -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix:rw -v /dev/shm:/dev/shm -v $HOME:/root/workdir deepnest:debian-bullseye`

Note:  
I use the option `-v /dev/shm:/dev/shm` to resolve this problem: *FATAL:memory.cc(22)] Out of memory. size=262144*

cmidgley, 2023-01-25:
I made a fork of Deepnest, starting with [Dogthemachine's fork](https://github.com/Dogthemachine/Deepnest) (it had some improvements that looked worthwhile), and then spent a while debugging how to get it running again considering Atom's shutdown and all the build issues above.  My only goals were to get it running again, simplify the build process, and document the prerequisites.  

I'm not in a position to support it or help with its future at this time.  And I've only done Windows (sorry), though some of the updated package dependencies may help for the other platforms (and the other changes might provide guidance on whatever else needs to be done).

But - if you want to build it for Windows, it's pretty easy now (though does depend on a lot of old packages)!  See the README for the steps necessary to build.  Most of the build itself is automated in `package.json`, but do pay careful attention to the prerequisites (I've tried to give guidance and links for all of them).

https://github.com/cmidgley/Deepnest

cmidgley, 2023-01-26:
Hi again - more improvements have been made in my [fork](https://github.com/cmidgley/deepnest) (see the readme for a list of changes, with some very minor bug fixes, removing a bunch of old files that were not used, and cleaned things up to remove several build prerequisites).  One thing, in particular, is I've tried to improve the multi-platform building but don't have access to a Mac or Linux at the moment.  If anyone wants to try it it would be great to know if it works on either/both of those platforms now.

bosd, 2023-04-16:
Maybe this personal repo can be converted to an organization. So more peopl can be assigned as a maintainer. Easier to hop on and off of collaborators. So @Jack000  wont have to manage the workload of this project on its own.

Trough the time there are quite a few forks with improvements. Would be great to have them merged in one central repo again.

cmidgley, 2023-04-16:
@bosd Feel free to create an org and fork/clone my fork.  Personally, I don't have the time right now to take on the ownership of managing this.

bosd, 2023-04-16:
@cmidgley  I went ahead and did the easy part.
Which is creating an organizaton.  https://github.com/deepnest-io

But it is empty now. First step would be to fill it with an repo.
I have not been in contact with @Jack000. So I don't know if he is willing to [transfer](https://docs.github.com/en/repositories/creating-and-managing-repositories/transferring-a-repository) this repo to the organization.
(So all PR's and issues and history will follow)
I invited @Jack000 to the owners role of the organization. So he has the ability to do so.
And if he wants to be involved, with the future of this project... he can..

As we don't know (if/when) the repo is transferred. I assume it is a good start to fill the organization with a fork of @cmidgley .  

I'd suggest to avoid a single point of failure. So have multiple owners and maintainers.
So anyone interested in collaborating in the organization let me know.
( I invited people who have shown interest in reviving this project as well)

As for contributions, my technical skills are not on par. But I can manage the organization.
Let's hope contributors come on board :smile:

DayTimeDreamer, 2024-10-18:
I have implemented DXF -> SVG yet using kabeja, but kabeja doesn't look capable of doing reverse conversion.

Could you tell me how to implement the conversion from SVG to DXF? I would need to know to implement offline.

ShaMan123, 2024-12-16:
@Dexus take a look

Dexus, 2024-12-17:
We will include the offline DXF2SVG and SVG2DXF conversion in our fork. However, there will only be a limited function. Online will still be available, as there are other ways to implement this.

Just for the information of those interested.

Dexus, 2024-12-21:
You can find the fork at github.com/deepnest-next/deepnest

The org github.com/deepnest-io was overtaken by an asshole by @abebeos and @lazaridiscom his alter ego account, which deleted now anything and resetting the contributions back to a version around v1.2.0 at his moment. And also kicked in the last days all members of the Organisation. 

I will rewrite in the future the complete app and make it ready for the future.

```

  "login": "abebeos",
  "id": 165260843,
  "created_at": "2024-03-28T08:34:11Z",
  "updated_at": "2024-12-12T17:04:50Z"

and

  "login": "lazaridiscom",
  "id": 62181287,
  "created_at": "2020-03-14T15:33:46Z",
  "updated_at": "2024-07-19T11:50:07Z"

```

I would not trust the deepnest-io org in any ways in the future as long there his accounts have ownership.