Jack000/Deepnest #124: Unable to define parts that belong together before sorting

IDFWT, 2021-04-18:
I've been trying to use Deepnest for nesting artwork, unfortunately there are pieces I can't define as being a part of another to avoid having it sorted elsewhere. 

[image]

For example, in the image above I would want to have Deepnest ensure those three pieces highlighted would be linked to each other. As of right now it is moving the leaf of the apple and a white piece I created near the characters mouth that's there for cutting purposes. Original image included if it helps any: 
[image]

jestelle, 2021-04-20:
I think this is conceivably possible by grouping elements in your design software (Illustrator, Inkscape, etc).
But I haven't personally found a consistent way to do it yet.

dorkmo, 2021-04-20:
the few times i needed a group, i drew a line connecting the two shapes, then deleted it after running deepnest

9swampy, 2021-09-04:
It seems like you're wanting to have parts cut out (the black lines - plus the green leaf) and the inner colourings (everything except the green) would be used to engrave? Is that correct?

The way the nest works is it extracts a representative polygon including an outer edge and any inner "holes" contained within the outer. That's what you're seeing in the list of parts. I've only worked with Dxfs and havn't seen anything that would suggest it has any concept of the colouring you're expecting it to include within the outer cut; if you're using Dxfs try using Svgs... but I can't confirm that would fare any better.

What it will do is use the original image, rotated, when it's doing the export (having used only the representative approximation during the actual nest to work out positionings). If you can force it to recognise your image as a single entity, perhaps as Dorkmo suggests, when you export it might reimpose the colouring detail.

PoetikDragon, 2022-04-08:
Has there been any solution for this yet?

ModischFabrications, 2022-04-24:
I want to use this tool to generate stickers. Tight nesting is great for optimal sheet usage, but this tool ignores group relations. Changing the code to honor (inkscape) groups would be fine for me, but seems unachievable without a new maintainer. 

Adding additional lines as @9swampy and @dorkmo described might work to connect logical areas, but it won't conserve colors, linewidth and others. 

Has someone found a workaround that conserves visual appearance?