Jack000/Deepnest #128: Optimize for Direction

Mopholo, 2021-05-20:
First really cool program

I used deep nest to layout parts for carbon fiber composite prepreg sheets, and it works great., not recutting where lines merge is a nice feature! 

One feature that would be nice is to be able to optimize for the direction of the layout on the sheet.  Fill width before extending the height of vise versa 

We unroll material, the width is fixed, the length is variable up to the length of the cutting table. We try to unroll as little material as possible the night before since it must be stored frozen before use and needs time to thaw. 

That means I want to optimize for width, and only add just enough height to pack all the parts in. 

Right now I just have a bunch of dxf 'sheets' exported each about 1" shorter than the next and start small, and work my way longer if necessary. 

I hope that makes sense, maybe there is already a way to do this, and I didn't understand it

Thanks

mikka8086, 2021-06-15:
It looks like "optimisation type" = "gravity" should achieve this. Check the preferences.

9swampy, 2021-07-07:
Gravity pulls to the left; just flip width and height, do your nest on a tall sheet instead of a wide one (or vice versa) then rotate the output.