Jack000/Deepnest #133: Missing libraries in Appimage for Linux

tuxcnc, 2021-09-26:
Hi.
On modern Linux (eg. Ubuntu Hirsue Hippo) you get very confusing error :
(deepnest:3808): Pango-ERROR **: 22:22:52.979: Harfbuzz version too old (1.2.7)
In fact, the libraries are too new, not too old.
Solution :
I was unpack the appimage, copied from <Ubuntu 18.04>/usr/lib/x86_64-linux-gnu/libpango* to <apimage>/usr/lib/*, packed again, and all works fine on Hirsue Hippo. 
The libpango files version is 1.0.so.0.4000.14 
Regards.

smartin015, 2021-12-09:
Tried this, but still received the same Harfbuzz version error:

```shell
/Deepnest-1.0.5-x86_64.AppImage --appimage-extract
cp /usr/lib/x86_64-linux-gnu/libpango* ./squashfs-root/usr/lib/
./appimagetool ./squashfs-root deepnest
./deepnest
```

```
(deepnest:2418861): Pango-ERROR **: 08:59:12.243: Harfbuzz version too old (1.2.7)

[1]    2418861 trace trap (core dumped)  ./deepnest
```

The files that were in my `squashfs-root/usr/lib`:
```
libappindicator.so.1  libindicator.so.7  libpango-1.0.so.0         libpangocairo-1.0.so.0         libpangoft2-1.0.so.0         libpangomm-1.4.so.1       libpangoxft-1.0.so.0         libXss.so.1
libgconf-2.so.4       libnotify.so.4     libpango-1.0.so.0.4800.2  libpangocairo-1.0.so.0.4800.2  libpangoft2-1.0.so.0.4800.2  libpangomm-1.4.so.1.0.30  libpangoxft-1.0.so.0.4800.2  libXtst.so.6
```
Running Ubuntu 21.04 (hirsuite): `Linux l2 5.11.0-40-generic #44-Ubuntu SMP Wed Oct 20 16:16:42 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux`

tuxcnc, 2021-12-09:
NOT cp /usr/lib/x86_64-linux-gnu/libpango* ./squashfs-root/usr/lib/
You must copy files from Ubuntu 18.04.

JamesRH, 2021-12-29:
This worked for me as of 2021-12-29 on Ubuntu 20.04.3 LTS x86_64 GNU/Linux

Get Deepnest and unpack 

    wget https://deepnest.io/Deepnest-1.0.5-linux.zip
    unzip Deepnest-1.0.5-linux.zip 
    Deepnest-1.0.5-linux/Deepnest-1.0.5-x86_64.AppImage --appimage-extract
    
get Ubuntu 18.04 libs and put in container dir

     for PKG in libpango-1.0-0_1.40.14-1ubuntu0.1_amd64.deb libpangocairo-1.0-0_1.40.14-1ubuntu0.1_amd64.deb libpangoft2-1.0-0_1.40.14-1ubuntu0.1_amd64.deb libpangoxft-1.0-0_1.40.14-1ubuntu0.1_amd64.deb
        do
        echo $PKG
        wget http://archive.ubuntu.com/ubuntu/pool/main/p/pango1.0/$PKG
        dpkg-deb -xv $PKG ./
        done
        
    cp ./usr/lib/x86_64-linux-gnu/libpango* squashfs-root/usr/lib/

This includes the following files:

        /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 -> libpango-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 -> libpangocairo-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 -> libpangoft2-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangoxft-1.0.so.0 -> libpangoxft-1.0.so.0.4000.14
        /usr/lib/x86_64-linux-gnu/libpangoxft-1.0.so.0.4000.14

rebuild AppImage and run

    wget "https://github.com/AppImage/AppImageKit/releases/download/continuous/appimagetool-x86_64.AppImage"
    chmod a+x appimagetool-x86_64.AppImage
    ./appimagetool-x86_64.AppImage ./squashfs-root deepnest

    ./deepnest

smartin015, 2022-01-14:
Thanks for sharing @JamesRH, this worked for me!

mattspeth, 2022-12-12:
Using Ubuntu 20.04.5 LTS 64x, same issue as above as of 2022-12-12.

@JamesRH solution above works well - thank you!