Jack000/Deepnest #135: Nested DXF exports duplicate layers

leeboy4130, 2021-11-21:
When I export the nest to dxf, there are duplicate layers of my shapes. When this dxf file goes to my CNC plasma program, it cuts the objects out twice since there are two duplicate layers on top of each other. The only way I know to fix this is to open the dxf file and delete each duplicate layer prior to sending to CNC plasma program. Am I the only one with this issue?

HaoLi111, 2022-10-30:
Same it would be nice if the surrounding layer is an actual layer with another color so that it would be easy to remove

WallyChal, 2024-10-16:
Is anyone looking into this issue?

bosd, 2024-10-16:
Maybe you will have more luck over at the community based fork

geimist, 2024-10-16:
> community based fork   

Which do you mean?

bosd, 2024-10-16:
https://github.com/deepnest-io/Deepnest

mattydboom, 2024-10-24:
Although that fork does seem much improved (for example; allows multiple dxf import!) I am still having the issue as described here.... the exported dxf files shows any curved lines with duplicate inboard/onboard lines. These are not needed and a pain to delete away initially.