Jack000/Deepnest #136: Fixed position & rotation of chosen shapes

dumblob, 2021-12-01:
It seems there is no way of "fixing" a position and rotation of manually selected objects and letting all the others get automatically positioned.

Do you think this would be possible to add?

cmidgley, 2023-01-26:
I know this is a rather old question - but I'm curious, why would you want to fix the position?  I understand rotation (to align with the material grain), but what is the use case for fixing a position?  Fixing rotation, by the way, is easy (set "Part rotations" in the settings pane to 1).

dumblob, 2023-01-26:
It is useful to help the algorithm in cases when the parts to be laid-out have too much varying sizes and thus one has a very few large parts and many mid-size parts. This is surprisingly common scenario and the algo does not a good job here.

In other words, the randomization of **initial** positioning (and rotation) of the algo seems to lead to local maxima more often than not (at least in an acceptable time frame). So the initialization has major effect on the end result and placing a few large shapes e.g. to corners of the available space could help the algo place the parts much better.

Yet in other words, human makes better estimations for initial conditions than random choice or some simplistic heuristics.

dumblob, 2023-01-26:
Another reason (perhaps even more important for SMEs & home-manufacturing) is to be friendly to our environment and to our budget :wink:. As this allows one to use low-quality material (which has certain places of different shapes stained/broken/scratched/...) or an already used material which has different cut holes in it etc.

cmidgley, 2023-01-26:
Thanks for the background.  Your first point seems to deal more with not being satisfied with the placement, especially on shorter runs.  It sounds like you have found conditions where you felt a better placement of some (perhaps the large shapes?) would have resulted in a denser nesting than it comes up with automatically.  If you happen to have any examples of this, that would be very helpful.  That seems like a candidate for algorithm improvement.

Your second point is very interesting as well, the idea of being able to work around material defects (or previously cut sections that you want to avoid).  One idea there might be to place shapes (at specific locations) that are marked as "do not use" and let nesting work around them.  The idea of setting a locked-down position for a shape would partially work, but that shape would still get cut out when you wouldn't really want that.  Additionally, the UI could render it differently (such as a cross-hatch pattern) to make it obvious what the intent is.

So it sounds like potentially three ideas here:

1) Look at the challenge you have with poorly placed items (for this, an example SVG or two would be needed)
2) Have a feature to identify areas on material that should not be allowed (wouldn't it be cool to use a laser-mounted camera, such as what LightBurn does, to help with that?!)
3) Offer the ability to specify shapes that should not be moved or rotated (or to limit how much rotation is allowed per shape, such as allowing only 180-degree rotations to maintain grain direction or disallowing rotation on a shape-by-shape basis).  

I want to be clear about my intentions.  I do not plan on adding features to Deepnest (though I do have a [fork](https://github.com/cmidgley/deepnest) that has fixed the build issues, at least for Windows; it currently has no release builds so you have to build it yourself).  Instead, I am looking at building a new nesting platform that is more sophisticated than Deepnest and was very interested in your issues to help refine the requirements.  Don't hold your breath - it is a big project that will take time (if it ever...!)  

I appreciate your time and guidance, and I'm open to hearing about other ideas or problems.  You can contact me here (via GitHub) or at my website (https://nucarve.com) - and I'm glad to share my email if you want (just don't want to put it on a public site).

Chris

dumblob, 2023-01-27:
@cmidgley ah, I see.

Well, if you are considering creating another tool which would also be open-source, then I am more than happy to help (not just use cases identification but also code reviews, architectural suggestions to save development time etc.).

I now took a look if I still have the "bad shape collections" which did not place nicely but I could not find them. I really just remember those were more geometrical L-shaped, T-shaped shapes (and other "perpendicular" shapes with low count of "tentackles") of larger sizes (larger in relation to the available material area - 50% or more of the largest material dimension).

dumblob, 2023-01-27:
Oh and btw. I would like to see placement which also tries to "share edges" of shapes. For such geometrical shapes I mentioned that could significantly (tens of percent) cut down the time to manufacture as well as the needed area (much less than tens of percent though).

cmidgley, 2023-01-27:
@dumblob Thanks for the offer to help.  I'll keep that in mind for sure, but it's a bit early now.  

For the "share edges" idea - did you try the "Optimization" setting?  It's supposed to manage the balance between space optimization and time optimization, though I've not verified if it really works or not.

dumblob, 2023-01-27:
> @dumblob Thanks for the offer to help. I'll keep that in mind for sure, but it's a bit early now.

Let me know if it ever materializes (either in this thread or send an email to my github nickname at gmail dot com).

> For the "share edges" idea - did you try the "Optimization" setting? It's supposed to manage the balance between space optimization and time optimization, though I've not verified if it really works or not.

I did not try this, thanks for suggestion! I just remember results and never saw a shared (part of an) edge.

cmidgley, 2023-02-02:
@dumblob I just discovered something interesting.  With Deepnest, when you mark an imported document as a Sheet the code does the correct outer and inner NFP analysis to find holes.  The parts within the sheet will maintain their X/Y position, and their interiors become ineligible for part placement.  This should address your desire for an exclusion region on stock, and gets pretty close to locking a part at a specific location as well.

dumblob, 2023-02-02:
Wow, that sounds super cool! I have to definitely try it.

Did you read code to discover it?

cmidgley, 2023-02-02:
Yep. Was looking at an import problem I’m having with a complex path in an SVG file and fell upon some code that looks at at inner NFP on stock (sheets) and that got me thinking.  I tried it and it seems to work.