Jack000/Deepnest #139: Feature Request: Import multiple DXFs at once.

piopio-eng, 2022-01-28:
It's annoying to have to do this one-by-one.

derekblankmccoy, 2022-01-30:
#97

Hillbillyrockr, 2022-03-30:
Agree this would be a awesome feature.

mmoriguela, 2022-04-21:
Is it possible to import multiple dxf's at once?

DavidWishengrad, 2022-05-07:
Yes, you can use netdxf right here on github to put all of your files in one file. It's a dll resource library in C#. It does not have a viewer, etc, but you don't really need that. You just have to write the code to open the the files in a list that you make and then clone everything into a new dxf file and then save that new file. It's all there for you if you any sort of coding skills in C#. It will be very easy. If you go this route, please acknowledge it only being easily possible with Daniel's work. Anyone can do this stuff now. That is a good thing.

cmidgley, 2023-01-26:
The fork by [Dogthemachine](https://github.com/Dogthemachine/Deepnest) has that feature, and some other enhancements as well.  I've [forked that fork](https://github.com/cmidgley/deepnest) to fix the build issues so it can be built (on Windows) but my fork doesn't have pre-built binaries so you may be better off using his (unless you want to build it on Windows...?)