Jack000/Deepnest #150: Nesting in loading

SimBro, 2022-09-07:
Hi
I installed the software (also on more laptop) but I can't get any nesting completed

The process gets stuck in the loading (the screen in attachment here)
[image]

Do you know why?
Thanks so much

Pinaute, 2022-09-07:
Hi, You have to add a sheet -> "0 sheet used"

SimBro, 2022-09-07:
Where do you mean ?
I flagged the parts, but I don't find the sheet dimension

[image]

SimBro, 2022-09-07:
Oh I understood my error

The flag is the sheet I have to use

Sorry eheh