Jack000/Deepnest #159: Hope export can keep bending lines layer

PandaBalls, 2023-03-23:
I need use laser cut machines to engrave the bending line, Fusion 360 Nesting can not export bending lines either.

Drider66, 2023-07-14:
Having the same issue. I start with a DWG on AutoCAD, first I tried leaving the bending lines as dotted one since I thought that was the issue. Unfortunately, I did changed bending lines to continuous one but this did not solved the issue. Hope someone could really help us with this. :(

bosd, 2023-08-30:
Thanks for your interest and feedback on deepnest.
However, this repo is no longer supported.
The author is no longer maintaining the repo or software.
There is a community based based edition available at:
https://github.com/deepnest-io/Deepnest

However, to move the deepnest forward we need more contributors / sponsors over there.