Jack000/Deepnest #161: Harfbuzz version too old (1.2.7)

Rybec, 2023-03-26:
This issue has been reported twice, once in Feb 2020 (#87) and once in Sept 2020 (#109).  Both times people suggested cheap hack workarounds to tide over the person who posted the issue over until the bug was fixed, but both times, the person who posted closed the issue, despite the fact that the issue _wasn't_ ever fixed.  It is now 2023, and this _fairly serious bug_ has existed for 3 years, and there's no open issue for it.

I'm not sure there's any point posting this, as this project appears to be completely dead, but just in case someone eventually picks it up, I'm going to repost those bug reports, so that maybe it will eventually get fixed.

I _would_ like to request that if this is _not_ going to be fixed, _please_ add a note to the website saying that the project is dead and if you aren't a power user who is comfortable compiling 3rd party libraries, it's not worth bothering (at least for Linux).  It's pretty rude to just let people waste their time downloading and installing it, just to find out that the project doesn't work out-of-the-box anymore (at least for Linux), when it still comes up near the top of searches and has a fairly nice web site that at least appears to be pretty recent.

Just for the record, I'm _not_ saying you should continue working on the project just for me.  If you don't have time to keep doing free labor to keep an open source project up to date, I _totally_ respect that.  My problem is when people aren't up front about it, and leave things lying around that make it look like the project is still going when it is so old that it doesn't even work on modern systems without a bunch of extra hacking.  I am grateful for the work you _did_ put into this, and I greatly appreciate it.  I would just have appreciated a warning, so I could decide what I was willing to do _before_ spending time and bandwidth on it.

Anyhow, I'm not looking for help.  I just wanted to make sure there was an open issue for this.  I'm going to try the cheap hack in #109.  I'll probably be fine with that (I _am_ a power user).  Honestly, I'm kind of surprised that this was packaged as an appImage but these libraries weren't included.  Isn't that the whole _point_ of appImage format?

bosd, 2023-04-20:
> I'm not sure there's any point posting this, as this project appears to be completely dead

Hi @Rybec True, This project seems dead, and no longer actively maintained by jack.
But still there is great interest in a program like this.

There is now a new organization with the goal of preventing this software to become abondonware.
As well as restarting a community to take over development.

As you indicated, one of the first goals is to get the program to run again on recent windows/linux versions and packages.

Please check out the repo at https://github.com/deepnest-io/Deepnest
Please post your issues and findings there.

Rybec, 2023-04-20:
Thanks for letting me know!  I did jump through the hoops to get the original running on Linux (putting old libraries in the executable directory), and it's working pretty well for me.

It's nice to see someone is working on it, though looking at the documentation, it looks like they aren't even trying to update the libraries and are only bothering to update other things for the Windows build at this point.  A bit disappointing, but at least it isn't completely forgotten, and maybe someone will eventually do some of the other stuff.

Again, thanks for letting me know!  I'll keep an eye on this.

bosd, 2023-04-20:
Well, the code in that repo is a fork of cmidgley's code. As he did some work to get it running on windows.
The documentation in the repo needs updating. (I did some quick changes).

I'm primarily on linux..
The old laptop on which is was running died on me.
Can you share your steps to get it running?

I've tried the docker version, but thats not working for me either. (Using podman on fedora).
Can you share some steps, to get it running?

The libraries do need some attention. I'm not keen on installing python 2.7 on a brand new machine just or this program. There are some forks, with more recent versions. (but they lack other improvements as there was no active central repo anymore)

>It's nice to see someone is working on it,

Well, don't be mistaken..
I'm the initiator of that repo. But it is not in my skillset to work on this kind of programs and languages.
cmidgley did a lot of work already, yet is not actively working on it at this moment.
There are some other members of the repo, but did not post yet..
But the repo it is only a few days old. let's hope more people can find their way there.

Rybec, 2023-04-21:
Oh, thanks for letting me know.  I should have realized that might be the case.  My appologies.

I followed the instructions in issue #109 in [this comment](https://github.com/Jack000/Deepnest/issues/109#issuecomment-710869498).

I don't remember if the wget command worked.  I seem to recall having to find that version of Pango somewhere else,  but I might be wrong.  (It might have been another thing I was working on, where I had to get an older version of some library and had to find it in some distro repository or whatever, because the link was broken.)

I'm using the appImage from this repository.  I put the Pango shared libraries (there are something like 12 of them)  in the directory with the appImage file, and I execute it from that directory.  If I understand correctly, this should work with the binary  as well as the appImage.

Thanks for explaining the situation.  I really wish I had time to help!  (Though, to be fair, JavaScript is one of least favorite languages...)  I couldn't believe such a useful project had just been abandoned, but then, I didn't do a very good job of looking for forks, so that's my fault.  I hope you are able to make some progress on this, even if you aren't able to do all of it on your own.

And thank you for organizing this.  I haven't got much value from the program yet (I'm pretty good at arranging things manually), but as I  gain more experience and want to do more complex things, it will be really useful, especially if it continues to work properly as technology advances and improves.