Jack000/Deepnest #164: I hate to suggest this, because I know I don't have the know how.

ruanhuishi, 2023-05-10:
I always encounter this error when running projects. Can you help me solve it?

              I hate to suggest this, because I know I don't have the know how.
Could someone please make a 'deepnest for rookies' youtube video.  Install
and get it going.
Other than being a 'handy' reference for reference, it might get some new
blood into helping
or at least keeping the project 'alive'.

I get all kinds of 'good ideas', but run out of steam before they are
implemented.

On Thu, Jun 17, 2021 at 2:36 PM 9swampy ***@***.***> wrote:

> Thx @kelyamany <https://github.com/kelyamany> those steps really helped;
> but for others trying same I can add more info
>
>    1. Download Boost 1.62 and unpack it to C:/boost (note this must be
>    same drive as the main code you're building; my code was on D so I needed
>    to put at D:/boost)
>    2. You need Python 2.7 (I got 2.7.18 because gyp(?) isn't compatible
>    with Python 3)
>    3. You need Visual Studio (or possibly the SDK; as @kelyamany
>    <https://github.com/kelyamany> I had VS on the box already)
>    4. npm install --target=1.4.3 --arch=x64 --dist-url=
>    https://atom.io/download/atom-shell" (at least I couldn't find any
>    1.4.8)
>    5. Run .\node_modules\.bin\electron-rebuild.cmd
>    6. Copy the contents of build/Release to minkowski/Release
>    7. npm run start or electron .
>
> That got me to a running application that executes the nest happily;
> unfortunately I have one bump - the preview graphic works fine but the view
> of the resultant nest doesn't show anything and it seems like the exported
> file isn't valid. Close but no cigar. Anyone got any advice to offer?
>
> On another note I also came across https://github.com/fel88/DeepNestPort
> which is a more actively supported port in C#.
>
> —
> You are receiving this because you are subscribed to this thread.
> Reply to this email directly, view it on GitHub
> <https://github.com/Jack000/Deepnest/issues/123#issuecomment-863511154>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/AAEUXERVF73QSUAHSD2A543TTJFE3ANCNFSM426BAFJQ>
> .
>

-- 
><> ... Jack

If you are not paying for something, you are not a consumer, you are the
product. - Chamath Palihapitiya

"Tell me and I forget. Teach me and I remember. Involve me and I learn." -
Ben Franklin

_Originally posted by @servant74 in https://github.com/Jack000/Deepnest/issues/123#issuecomment-863530980_

ruanhuishi, 2023-05-10:
I always encounter this error when running projects. Can you help me solve it?
[image]

ruanhuishi, 2023-05-11:
The above issues have been resolved, but new issues have emerged.

 fatal error C1083: 无法打开包括文件: “boost/polygon/polygon.hpp”: No such file or directory (编译源文件 ..\addon.cc) [E:\3-code\Deepnest\Dn4\Deepnest\build\addon.vcxproj]
fatal error C1083: 无法打开包括文件: “boost/polygon/polygon.hpp”: No such file or directory (编译源文件 ..\minkowski.cc) [E:\3-code\Deepnest\Dn4\Deepnest\build\addon.vcxproj]

gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\Professional\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (E:\3-code\Deepnest\Dn4\Deepnest\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at emitTwo (events.js:126:13)
gyp ERR! stack     at ChildProcess.emit (events.js:214:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
gyp ERR! System Windows_NT 10.0.19042
gyp ERR! command "E:\\0-step\\nvm-setup\\nvm\\v8.12.0\\node.exe" "E:\\3-code\\Deepnest\\Dn4\\Deepnest\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
gyp ERR! cwd E:\3-code\Deepnest\Dn4\Deepnest
gyp ERR! node -v v8.12.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! Deepnest.io@1.0.5-db install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the Deepnest.io@1.0.5-db install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     D:\nodejs\node_cache\_logs\2023-05-11T02_19_05_320Z-debug.log
[image]

ruanhuishi, 2023-05-11:
[image]

I have resolved the above issues; I am a novice in this project and I am very sorry

ruanhuishi, 2023-05-11:
[image]

running  ！

but   start  nesting  is  bug

abebeos, 2023-05-11:
Please visit https://github.com/deepnest-io/Deepnest

This project here is unmaintained (you can close your issue)