Jack000/Deepnest #169: Deepnest keeps messing up my dxf files imported from solidworks

KRGrifon, 2023-09-21:
Please help, I dont know what to do here...
Some of the parts get deformed when I import them into the program, but theyre fine when I open them as dxf in SW...
Take a look:

[image]
[image]

KRGrifon, 2023-09-21:
Also, my export button doesnt produce a dxf file. When saving, it just says 'all files' and when its saved, it wont show up as dxf and I cant open it...

bosd, 2023-09-21:
Maybe you can take a look at the community based fork. Could be that your issues are fixed.
https://github.com/deepnest-io/Deepnest