Jack000/Deepnest #170: Deepest cannot replicate a professional layout

Ewgeniy77, 2023-10-10:
Hello dear developers. I have a layout made in a professional program, the size is 4026x1380 mm. There are 52 details. I created a rectangle of the same size and marked it as a page. I sent the details to the "Start nest" layout. For two days with different installations, I could not get the same result, the maximum Deepest was able to accommodate 43 parts out of 52. Can you advise something in this situation? Thank you.

[image]
[image]
[image]
[image]
[image]

Ewgeniy77, 2023-10-10:
[image]

radevgit, 2023-10-11:
Let me ask you, why do you believe that two different products with completely different algorithms will produce the same quality results? The problem is NP complete (meaning very hard) and the produced results are usually not optimal, but "the best effort for given algorithm". There is no proof of optimality or any guarantee of the quality of the result. My guess is, that the "professional program" is using large set of placement heuristics and proprietary algos. The deepnest use some sort of genetic algo. See the references here: [https://github.com/Jack000/SVGnest](https://github.com/Jack000/SVGnest)

Ewgeniy77, 2023-10-11:
Thank you for your reply. I did not write that I am not satisfied with the program. I asked if it was possible to make adjustments for better placement to get as close to the sample as possible. I see in the title of the program it says "industrial nesting", so I thought it might be possible.

chetan-SMRI, 2023-10-26:
> Thank you for your reply. I did not write that I am not satisfied with the program. I asked if it was possible to make adjustments for better placement to get as close to the sample as possible. I see in the title of the program it says "industrial nesting", so I thought it might be possible.

you are right, i also think that this doesnt produce the most accurate results possible, other paid tools use different algos but produce industry standard result