Jack000/Deepnest #191: Holes in parts are not staying within the part.

StevenMarston, 2024-08-05:
First off, let me say this is an amazing piece of software!!

I have parts to be nested that have holes in specific locations.  I've grouped the perimeter of the part and its hole prior to exporting from LightBurn. Inkscape and other software shows the holes are in place as needed.

However, upon importing the SVG or DXF into DeepNest,  the holes become disconnected and are placed outside of the part. (Does the same for SVGnest).

Need the holes to stay in their location as having to put them back in place after the parts have been rotated and moved is far too much trouble. Doable, yes, but difficult.

Dexus, 2024-11-02:
DXF works fine! its so only a svg problem

StevenMarston, 2024-11-03:
Thanks for the response. However, as  I stated earlier,  this issue is also occurring with the DXF version of the output.
I just kept trying it over and over with the same file  and I eventually got a usable nested file.

Dexus, 2024-11-03:
Then fix the Issue and make all forms a path. So you should not have the problem.

StevenMarston, 2024-11-03:
If I knew where the issue was with the exported file, I wouldn't be asking for help here.

Making "all forms a path" is found where? Before I export? After I export (and in which recommended software?)

Seriously confused by your intended verbiage of help.

Could you explain it better in a "dumbed-down" version for me, please?

**REPEAT**
_BTW, did I say I really appreciate this project?
Probably more than you know. Thanks to all the developers!_

Dexus, 2024-11-10:
@StevenMarston 
open the SVG or DXF in Inkscape and select all elements, then CTLR + SHIFT + C (Path>Object to Path) 
and export it again.

Dexus, 2025-03-01:
Check www.deepnest.net for new release of v1.5.0 from our deepnest-next fork. 

With enabled setting svg normalizer all parts stay with the other parts.

Dexus, 2025-05-17:
This should now supported with the upcoming release of v1.5.6 at www.deepnest.net