Jack000/Deepnest #196: Errorcode 522

Tigger86, 2025-03-22:
I receive error code 522 when trying to upload a DXF file. What could be the cause? I have already reinstalled the software, but unfortunately, it didn’t help.

alexisdal, 2025-03-23:
apparently the source code replies on a internal convertion webservice hosted on 

http://convert.deepnest.io

source:  https://github.com/Jack000/Deepnest/blob/2fb10513a30681971dcc991c528fa0738a2c0c76/main/index.html#L88

however, this internal website replies with a 522 HTTP error code (some sort of a timeout). I suspect the thing has been down for a long time ?

[image]

when I look at the apparently community driven alternative
https://github.com/deepnest-next/deepnest/blob/49a6aafa70d222d4dc73142c430d98c1c0e27285/tests/index.spec.ts#L102
it seems to rely on a similar approach but another hostname
https://converter.deepnest.app/convert
and this one does seem to respond...

when I try to load the SVG-based file on cambam (I tried different file converstion service online) => but the scale in cambam is completely proken + I lost all the holes in the part. it's a disaster

smells like bye-bye deepest.io 😭

bosd, 2025-03-24:
@Dexus Can you take a look here?

Dexus, 2025-03-24:
You can all switch to www.deepnest.net / https://github.com/deepnest-next/deepnest.

I already have a working converter, not much tested yet, but created for my own files and needs and found it good so far. But if you find any bugs that should be fixed, I'm here and actively working on deepnest-next.

Best regards, Josef

Tigger86, 2025-03-24:
unfortunately still an error. i have changed to version 1.5.0. tried to upload a dxf file... unfortunately i get the message : could not contact file conversion server.

Dexus, 2025-03-24:
@Tigger86 please ready all, inclusive my link to the Issue: https://github.com/deepnest-next/deepnest/issues/71#issuecomment-2744963030 in this comment I describe how to change the current converter API from your current settings - so you dont have to reset them. Same for Presets - if you had one from our version of deepnest-next.

Tigger86, 2025-03-24:
maybe i'm being too stupid... but i have changed the file... but still i get the message`{
  "units": "mm",
  "scale": 72,
  "spacing": 0,
  "curveTolerance": 0.72,
  "rotations": 4,
  "threads": 4,
  "populationSize": 10,
  "mutationRate": 10,
  "placementType": "box",
  "mergeLines": true,
  "timeRatio": 0.5,
  "simplify": false,
  "dxfImportScale": "1",
  "dxfExportScale": "72",
  "endpointTolerance": 0.36,
  "conversionServer": "https://converter.deepnest.app/convert",
  "useSvgPreProcessor": false,
  "useQuantityFromFileName": false,
  "exportWithSheetBoundboarders": false,
  "exportWithSheetsSpace": false,
  "exportWithSheetsSpaceValue": 0.3937007874015748
}`

Dexus, 2025-03-24:
I don't see any request to the server, so it can't be true. It works here only to change the file and restart the app. 

please try on windows:
```
# powershell
$env:deepnest_debug=1
#start deepnest from powershell
.\deepnest-v1.5.0.exe
```

ot should open a developer console, please look to the " console " and also to the " network " tab, copy the console to file and create from the network tab the HAR file, and attach it here, so I can check it.

Tigger86, 2025-03-24:
i hope i did it right.... 

[-1742848862059.log](https://github.com/user-attachments/files/19438109/-1742848862059.log)

Tigger86, 2025-03-24:
when i import an svg file it works

Dexus, 2025-03-24:
Thx will check it tomorrow

Dexus, 2025-03-25:
Thanks, for letting me know that it was not working, my tests run against a different config as the prod. Thats why it did not work in prod. Now it should fixed, with the change.

Tigger86, 2025-03-25:
you're welcome..  it could have been my mistake

Tigger86, 2025-03-25:
sorry i found another problem... 
I have updated to v.1.5.1... 
what I find positive is that you can now upload several dxf files at once.

the scale of the dxf is unfortunately not correct. so if i upload a dxf that is 100x100 mm, then it is 2540mm in deepnest... and it is similar when i export it as a dxf.... . if i export it as svg. then it works.

Dexus, 2025-03-25:
Can you please provide a test file for me? And Share the current used settings? On windows `%APPDATA%/deepnest/` the `settings.json`

Dexus, 2025-03-25:
@Tigger86 thanks, I have just review you settings, I have changed the default unit to inch for the converter and when I set the import dxd units to points, I get 100mm by 96dpi, also If i set the export to inch at 96dpi i get also 100mm. 

It would be so nice, if the Units would match..., I will check this. Please open an issue in our repo. Then i can track it. I've no watch on this repo so, it would get lost in my notifications.

Tigger86, 2025-03-25:
done.

RainerPri, 2025-03-27:
Is there a finished compiled app for MacOs?

I have found the installation instruction but i have no expirience with that.
What i have to do on MacOs?

I have installed XCode alllready!

1.st brew install boost

and then?

The building command's?
git clone https://github.com/deepnest-next/deepnest
cd deepnest
npm install
npm run build
npm run start

How i can made then an *.app file?

With that comman?
npm run dist

Thank you and sorry for the stubid questions.

Dexus, 2025-08-30:
If someone else could do it other than Jack, it would probably be that.